riscv/riscv-debug-spec #353: What states are possible while a hart is in reset?

jhauser-us, 2018-09-07:
The DM spec should say explicitly what states may be reported for a hart while being held in reset or while "coming out of reset".  Currently, in section 3.2, Reset Control, we have
> The system may take an arbitrarily long time to come out of reset, as reported by `allunavail`, `anyunavail`, or other implementation specific indicators.

I read this to imply that, while a hart is in reset, it should be reported as *unavailable*, unless there is some other "implementation-specific indicator" of reset.  If that's the intention, the spec would benefit from saying so explicitly, i.e., harts being reset are reported as *unavailable*, unless reset is indicated by some other, implementation-specific means.

Furthermore, I believe the current spec is not clear about what other states, if any, may be indicated for a hart while in reset.  Can or should a DM report *halted* or *running* for any harts being reset?  For example, if a halted hart gets reset, should (or may) the *halted* state remain asserted during reset?

Please note the interaction of this issue with #351.

Lastly, is there any possibility of dropping the option of an "implementation-specific indicator" for reset?  Even if a DM has an extra, nonstandard indicator, I would like for every DM to consistently report the state of harts in reset, whether that's as *unavailable* or some other agreed state.

jhauser-us, 2018-09-11:
The discussion for this issue seems to be occurring elsewhere, which I think is unfortunate.  For the benefit of anyone who might look here, some desire has been expressed to allow harts in reset to report a state other than *unavailable*.  Under #355, Tim Newsome wrote:
> The best solution is to say that while held in reset, harts may report being unavailable (no abstract commands possible) or running (abstract commands are possible). Running then means something like "doing what the hart would do without the debugger interfering."

I haven't seen anyone arguing yet for allowing harts in reset to report *halted*.  Does anyone believe that one should be possible also?

(From issue #351, it appears to be agreed that *unavailable*, *running*, and *halted* are always mutually exclusive.)

mwachs5, 2018-09-11:
[This comment isn't really suggesting anything, just running through the thought experiment]:

> I haven't seen anyone arguing yet for allowing harts in reset to report halted. Does anyone believe that one should be possible also?

If a hart was halted, then (unknown to the debugger) got reset and didn't halt out of reset, then it would be confusing to report it as Halted because the debug module wouldn't actually have control of it. But, the debugger could observe the 'hasreset' and deduce what happened. Still, not optimal situation.

Note that avoiding this scenario was why we added the `hasreset` indicator and the `setresethaltreq` control. I agree it does create an implicit, undefined state "maybe running but reporting halted" if you don't use those controls and indicator.

jhauser-us, 2018-09-11:
I think we can agree that a hart's state (*running*, *halted*, *unavailable*) implies that a certain set of functions is or isn't available to a debugger for controlling and accessing that hart.  Moreover, a debugger that believes it is in control of a halted hart has reason to want to be informed should that hart get reset.  (The situation is arguably different when a running hart is reset, since a running hart can be said not to be under the debugger's direct control.)

I believe it's reasonable to decree that, regardless of implementation, reset is assumed *in principle* to be incompatible with the debug functionality ascribed to the *halted* state.  Therefore, if a hart that was halted gets reset, the state during reset must be *unavailable*, because it can't be any of the other choices.  If the edits proposed in #322 are adopted, this has the side effect of requring that, for a halted hart, any active abstract command is terminated by reset (because the state becomes *unavailable*, however briefly).

If anyone wants to argue that a DM might be able to support all the same abstract commands for a hart in reset as for a halted hart, I say I don't even know what "reset" means for that implementation.  It sounds like in reality the hart was briefly reset and is then being held in the halted state, although this state is being called "reset" for some reason.  Rather than adapt the RISC-V debug spec to accept this arrangement, I would suggest adjusting the implementation's model to better reflect reality.

jhauser-us, 2018-09-11:
Megan Wachs wrote:
> If a hart was halted, then (unknown to the debugger) got reset and didn't halt out of reset, ...

I thought that wasn't allowed.  I've opened another issue, #357.

mwachs5, 2018-09-11:
Yeah, good point. That's not really a valid scenario.

commit b524a3e0 (2018-09-07): While held in reset, harts are unavailable.

Fixes #353.
commit fe6f04d8 (2018-09-07): Remove "impl specific [reset] indicators"

They don't supersede the unavailable status, and their mention is
confusing. (They were added in ce1b3593.)

Fixes #353.
commit 84ec8a50 (2018-09-18): Harts can be in exactly 1 of 4 states. (#354)

* Harts can be in exactly 1 of 4 states.

Fixes #351.

* Grammar fixes.

* While held in reset, harts are unavailable.

Fixes #353.

* Remove "impl specific [reset] indicators"

They don't supersede the unavailable status, and their mention is
confusing. (They were added in ce1b3593.)

Fixes #353.

* Unavailable harts can't execute abstract commands.

Being in reset doesn't mean you have to be unavailable. #355

* Clarify reset states.

* Abstract behavior when hart is/becomes unavailable

Just a rewording that (hopefully) makes more sense.
closed 2018-09-18 by timsifive