riscv/riscv-debug-spec #355: Clarify whether unavailable harts may execute abstract commands

jhauser-us, 2018-09-08:
The section on Abstract Commands (currently 3.5) discusses the execution of commands for harts that are either halted or running, but says nothing about unavailable harts.  May a DM support the execution of abstract commands for a hart that's reported as *unavailable*?  If so, under what conditions can and should the document actually define the behavior of such commands?

I would've assumed that unavailable harts can't execute commands, but Megan Wachs has suggested otherwise (#322):
> but maybe you want to be able to read registers in a hart while it's in reset...

and also:
> If an implementation supports a sideband read of a register, then that may and should work if the hart is in reset, not just give an error. This is useful functionality in general.

I've created this issue to attempt to resolve the question.

Realistically speaking, I believe it'll be rare, at best, that harts in reset are physically able to execute abstract commands, even something as simple as reading/writing the GPRs.  I personally know of no existing cores---of any kind---that would support such a thing.  And when a hart is unavailable for reasons other than reset (powered down, for example), it seems even less likely that abstract commands can be supported.

But even if/when they might be physically supportable, I question the practical usefulness of executing abstract commands for unavailable harts.  For a hart in reset, in particular, some of the most interesting information such as the PC value will typically have been changed by reset, so the picture you can get of the hart's state from before reset is fractured and potentially misleading if not properly understood.  And if the hart is set to halt on reset, then whatever information you can gather this way should be equally available once the hart comes out of reset.  You only need to wait.

Given these realities, I recommend for simplicity's sake that the document say that unavailable harts can't execute abstract commands, and that `cmderr` must be set to 4.

I welcome any "alternative facts" I overlooked.

mwachs5, 2018-09-10:
> Realistically speaking, I believe it'll be rare, at best, that harts in reset are physically able to execute abstract commands, even something as simple as reading/writing the GPRs. 

The main difference of the "debug RAM" of debug spec v11 and the "abstract command" of v13 was to remove the idea that harts have to "execute" anything at all. That's just an implementation detail. A hart is an architectural concept, and in theory the debug module could peek/poke at the registers at any time using abstract commands, regardless of whether the hart's clock is running, reset is asserted, it's in some low-power state, or whatever. Conceptually, it's the debug module which executes abstract commands, not the hart.

Further, the abstract command interface allows "custom" registers (not currently assigned to GPR/CSR/FPRs). Implementers may want to allow reading whatever those addresses map to even when the hart is clockgated/powered down/in reset. Perhaps that's even what they were added for. Some architectures actually dump state when reset is asserted for debugging purposes. Sure, you could query it when reset is de-asserted again, but must we forbid it?

But I think this is mixing unavailable and reset. I guess the question is, what is the point of reporting a hart as unavailable if you're allowed to execute an abstract command while it's selected? I agree with this that you shouldn't be able to execute an abstract command when the hart selected is unavailable. But I don't think that is *because*  that hart is in reset, or whatever. Unavailable *means* "you can't execute an abstract command, go read the docs of your design for why". I don't think we should assign a lot of physical meaning to what unavailable means.

timsifive, 2018-09-10:
#354 is relevant to this also.

I agree with @mwachs5 that it's unnecessary to limit abstract commands when harts are in reset. It doesn't make sense in the rocket implementation, but it could make sense otherwise. Allowing abstract commands while unavailable is weird and counter-intuitive. The best solution is to say that while held in reset, harts may report being unavailable (no abstract commands possible) or running (abstract commands are possible). Running then means something like "doing what the hart would do without the debugger interfering."

mwachs5, 2018-09-10:
FWIW:

> It doesn't make sense in the rocket implementation,

The rocket-chip debug module does now support "Custom" register access: https://github.com/freechipsproject/rocket-chip/pull/1606

It's completely undefined (since rocket-chip is a library anyone can use) what these would hook up to, but they could hook up to something that is still interesting when the selected hart is in reset.

Anyway, I think we are all in agreement that "unavailable harts can't execute abstract commands". Just a little unclear what happens then if a hart is/becomes unavailable while a command is executing. And the question of what makes a hart unavailable is a separate PR/issue.

mwachs5, 2018-10-01:
I think we resolved this @timsifive , but I lost track of which PR did...

mwachs5, 2018-10-01:
Addressed by #322

closed 2018-10-01 by mwachs5