riscv/riscv-debug-spec #357: Do halted harts remain halted after reset?

jhauser-us, 2018-09-11:
In section 3.2, Reset Control, we have:
> The halt state of harts should be maintained across system reset provided that dmactive is 1, ....

Does "halt state" here refer to the fact that a hart is halted, or does it mean the halt-on-reset option?  In the first interpretation, that would mean that all harts that were halted before system reset will be halted again after reset, even if their halt-on-reset options weren't enabled.  (Presumably, any harts that have the hart-on-reset option enabled will also be halted after system reset, but that's a separate matter.)

If the correct interpretation is the second one (harts are halted after system reset only if their halt-on-reset option is set), I believe the current wording is misleading.

On the other hand, if the correct interpretation is the first one (halted harts remain halted after system reset, in addition to any with the halt-on-reset option set), then does this property of "remains halted" apply also to resets of selected harts by `dmcontrol.hartreset`, or to resets that occur by outside means?  If so, this needs to be stated in the document.  If not, why not?

(FWIW, I favor keeping halted harts halted after reset.  Otherwise, I believe a debugger will regularly need to set and clear the halt-on-reset option to ensure this is the case.  Making the behavior automatic will help reduce the traffic required between a debugger and the DM.)

mwachs5, 2018-09-11:
Practically, if a debugger sets haltreq and doesn't clear it, in my mind it's the same thing... the hart will halt again out of reset. The halt-on-reset option is also useful in the case you just want to grab control of a hart straight out of reset (that you may not know why it's being reset), but dont actually want it halted now.

I guess what that sentence really means is the "haltreq" bits of the hart, not whether it's actually halted or not, if that makes sense.

jhauser-us, 2018-09-11:
> I guess what that sentence really means is the "haltreq" bits of the hart, not whether it's actually halted or not, if that makes sense.

That's one possibility, but it's not the same thing as my first interpretation, because harts may get halted for other reasons, such as executing EBREAK or due to a trigger.  It sounds like you're saying that if a hart was halted by `dmcontrol.haltreq`, it won't resume running after reset, but otherwise it does unless the halt-on-reset option is set.  I doubt that's what you really want.

mwachs5, 2018-09-11:
Good point. It might not be what we WANT, but not sure how to spec it in a practical way to implement. I think there is a baked in assumption in the spec that once a hart is halted, it's under debugger control. That goes out the window when considering the spurious, non-debugger controlled reset.

ernie-sifive, 2019-01-08:
I agree with @timsifive.  I don't believe a hart should stay halted after a reset unless its halt-on-reset is set.  That's the purpose of this feature - to specify which harts you want halted following a reset.  A hart's state prior to the reset should not matter.

PR #504 (2019-11-26): Halt state may not be preserved across reset.
It's not necessary to have this requirement, since there are other
mechanisms to ensure harts halt out of reset if that's what the debugger
wants.

Fixes #357.
commit 349c8264 (2019-11-26): Halt state may not be preserved across reset. (#504)

* Halt state may not be preserved across reset.

It's not necessary to have this requirement, since there are other
mechanisms to ensure harts halt out of reset if that's what the debugger
wants.

Fixes #357.

* Slight rephrasing.
closed 2019-11-26 by timsifive