riscv/riscv-debug-spec #359: Suggest changing haltreq and resumereq to be W1, not W

jhauser-us, 2018-09-11:
This may be a long-shot, but I really believe the `haltreq` and `resumereq` bits of `dmcontrol` would be better specified as W1 instead of W.

After submitting a halt request, a debugger could be prohibited from making a resume request on the same hart until the hart is confirmed to be halted.

For the opposite case, a debugger should always be permitted to make a halt request.  If a hart has a pending resume request that is not yet satisfied, an incoming halt request cancels the pending resume.  A debugger can observe `allresumeack`/`anyresumeack` to learn whether a halt request acted soon enough to prevent any resumption of execution.

timsifive, 2018-09-15:
The current definition allows `haltreq` and `resumereq` to be W1 if you like, but if you want to get fancy it's possible to cancel request. I prefer not to add more requirements on an implementation as to what happens in cases where the debugger changes its mind. Either debugging is working fine and halting/resuming is fast and it's irrelevant, or debugging is not working fine and reset is probably the only real recourse in any case.

mwachs5, 2018-10-01:
addressed by #372