riscv/riscv-debug-spec #374: Run control behavior

bdwyatt, 2018-09-28:
I'm trying to reconcile the description in section 3.5 Run Control, with the dmcontrol description and PR #321 comments.

My interpretation of the spec is that the expectation for a halt request goes like this:

1. Debugger writes 1 to haltreq in dmcontrol
2. Debug Module asserts the haltreq signal to the selected hart(s)
3. Selected harts, if not already halted: halt, clear running signal, and assert halted signal
4. Debugger writes 0 to haltreq
5. DM clears the haltreq signal to the selected harts

However if I read the comments associated with PR #321, it seems the intent for this process differs slightly:

1. Debugger writes 1 to haltreq in dmcontrol
2. Debug Module asserts the haltreq signal to the selected hart(s)
3. Selected harts, if not already halted: halt, clear running signal, and assert halted signal
4. Debug Module clears the haltreq signal in response to the hart halting

Resume requests, I would expect to work similarly.

Spec Version:
1. Debugger writes 1 to resumereq in dmcontrol
2. Debug Module asserts resumereq signal to the selected hart(s)
3. Selected harts, if not already running: resume, clear halted signal, assert running and resume_ack signals
4. Debugger writes 0 to resumereq
5. DM clears resumereq signal to the selected harts
6. Selected harts clear their resume_ack signal

PR #321 comment version:
1. Debugger writes 1 to resumereq in dmcontrol
2. Debug Module asserts resumereq signal to the selected hart(s)
3. Selected harts, if not already running: resume, clear halted signal, assert running and resume_ack signals
4. DM clears resumereq signal in response to the running/resume_ack going high
5. Selected harts clear their resume_ack signal in response to resumereq being cleared

Basically, is it the Debug Module or the Debugger's responsibility to clear halt/resume requests? Or is there something else I'm missing?

Additionally, the spec has clearly defined behavior for how a hart should drive its halted/running/resume_ack signals in response to halt and resume requests. However it's less clear on what should be expected in the case of a resume request being quickly followed by a halt event generated by single step, ebreak, etc. Essentially it's not clear which has higher priority: showing that the core has halted, or making sure that the debugger can see that the resume happened. I suppose there's also potential to show that both the resume happened and we've since halted, given the presence of of the resume_ack in addition to running. Something along the lines of if a halt occurs before resume_ack is cleared: running is cleared, halted is asserted, but resume_ack remains high until acknowledged.

timsifive, 2018-09-28:
In your description you're using the same name for an internal haltreq signal, and the haltreq bit that the debugger writes. Only the latter is covered by the specification. What happens to the internal one is between the DM and its hart.

The definition of the haltreq says:
> 1: Causes the currently selected harts to halt, if they are currently running.
You could implement this by having the DM pulse an internal haltreq signal when it is written, but that is an implementation detail.

The same is true for resumereq:
> 1: Causes the currently selected harts to resume once, if they are currently halted.

The intent is for the action to happen when the write happens. The bit doesn't persist beyond that. When the bit is written 1, an action happens. When the bit is written 0, a different action (possibly canceling an outstanding request) happens.

Does that clear things up? What could we write to make the spec more clear?

timsifive, 2018-09-28:
While I think I've accurately summarized the current state of the spec, this might change for haltreq in the next few days.

bdwyatt, 2018-10-01:
I believe part of the issue is that the dmcontrol register halt/resume request fields are not the sole source of information in the spec about run control. There is an entire section describing how this should work, which already has some expectations for how signaling between the DM and hart should work:

> 3.5 Run Control
For every hart, the Debug Module contains 4 conceptual bits of state: halt request, resume request,
halt-on-reset request, and hart reset. (The hart reset and halt-on-reset request bits are optional.)
These bits all reset to 0. A debugger can write one of them at a time for the currently selected harts
through haltreq, resumereq, setresethaltreq/clrresethaltreq and hartreset in dmcontrol. In addition
the DM receives halted, running, and resume ack signals from each hart.

> When a running hart receives a halt request, it responds by halting, deasserting its running signal,
and asserting its halted signal. The halted signals of all selected harts are reflected in the allhalted
and anyhalted bits. haltreq is ignored by halted harts.

> When a halted hart receives a resume request, it responds by resuming, clearing its halted signal,
and asserting its running signal and resume ack signals. **The resume ack signal is lowered when the
resume request is deasserted.** These status signals of all selected harts are reflected in allresumeack,
anyresumeack, allrunning, and anyrunning. resumereq is ignored by running harts.

> When halt or resume is requested, a hart must respond in less than one second, unless it is unavail-
able. (How this is implemented is not further specified. A few clock cycles will be a more typical
latency).

> The DM can implement optional halt-on-reset bits for each hart, which it indicates by setting
hasresethaltreq to 1. This means the DM implements the setresethaltreq and clrresethaltreq bits.
Writing 1 to setresethaltreq sets the halt-on-reset request bit for each selected hart. When a hart’s
halt-on-reset request bit is set, the hart will immediately enter debug mode on the next deassertion
of its reset. This is true regardless of whether or not the hart’s reset is caused by the DM’s
ndmreset/hartreset signals. The hart’s halt-on-reset request bit remains set until cleared by the
debugger writing 1 to clrresethaltreq while the hart is selected, or by debug module reset.

My suggestion would be to update the language in this section to present a more complete picture of debugger, debug module, and hart interaction required for run control.

If halt requests are going to become stateful again, per PR #378, that will impact the language used. However, here's a rough attempt at updating the halt request paragraphs from section 3.5.

> To initiate a halt request, the debugger selects the hart(s) to halt and writes a 1 to haltreq in dmcontrol. The DM then asserts the request to the selected harts, which is cleared on a per hart basis as harts signal that they have halted. When a running hart receives the halt request, it responds by halting, deasserting its running signal, and asserting its halted signal. The halted signals of all selected harts are reflected in the allhalted and anyhalted bits of dmstatus. haltreq is ignored by halted harts. Optionally, if the debugger writes a 0 to haltreq before the hart has halted, the outstanding halt request may be canceled.

Resume would need a similar update.

The other piece of information missing from the current spec is the priority of status from the hart. For example, the single step case, what should a hart show on its halted, running and resume ack signals after a step occurs? Is it acceptable for the debugger to potentially see no indication the resume happened?

A first pass at a potential solution to those questions would be:

> When a hart halts before resume ack has been cleared the hart shall assert halted and resume ack while deasserting running. Indicating that the hart did resume, but has since halted due to a source other than halt request, such as single step.

That would resolve one issue I've seen, where the debugger has one expectation for the contents of dmstatus after a single step, but the DM was implemented with a different expectation.

timsifive, 2018-10-01:
I've updated #378, hopefully addressing your concerns. Could you take a
look?

As for the single step example, I believe the behavior you want naturally
rolls out of the current definition. It certainly works that way with
existing OpenOCD.

closed 2018-10-03 by bdwyatt