riscv/riscv-debug-spec #395: is dmstatus.version readable when dmactive is 0?

timsifive, 2018-10-26:
@gravikumar92 raised the following problem in https://github.com/riscv/riscv-openocd/issues/316:

While dmactive is 0, the spec says:
> The module’s state, including authentication mechanism, takes its reset values (the dmactive bit is the only bit which can be written to something other than its reset value).
and also:
> Implementations may pay attention to this bit to further aid debugging, for example by preventing the Debug Module from being power gated while debugging is active.

The former implies that dmstatus.version is readable, but the latter implies it might not be. Currently OpenOCD assumes the version is always available, and it uses this to automatically detect 0.11 vs. 0.13. Since 0.11 doesn't have a dmactive bit automatic detection gets trickier (but probably still doable) if we can't assume the version is always accessible.

timsifive, 2018-10-26:
@gravikumar92, I assume you have an implementation that does something such that dmstatus.version is not accessible until dmactive is 1. How much more complicated does your design get if there is a requirement to always have dmstatus.version accessible? What other costs are there to that?

gravikumar92, 2018-10-31:
I have to check with the design team. As per our understanding, dmactive should be 1(init sequence) to read dmstatus register.

aswaterman, 2018-11-01:
@gravikumar92 IMO the behavior of reading dmstatus should be unspecified when dmactive=0.

timsifive, 2018-11-08:
@aswaterman Even the version field, which should be a constant and isn't dependent on any system state? Is this because it's expensive to implement a register returning a constant when dmactive=0, some philosophical issue, or something else?

timsifive, 2018-11-08:
The dmactive bit in 0.13 holds the place of fullreset in 0.11. Always writing dmactive and then checking version would result in resetting 0.11 systems.

mwachs5, 2018-11-09:
I think that you shouldn’t necessarily be able to connect to a system that
has already powered itself down or clock gated or whatever dmactive is
going to prevent.

mwachs5, 2018-11-26:
Ok, it seems like we would perhaps want an exception for reading DMSTATUS (or even only DMSTATUS.version?) even when dmactive=0? But that's not a strictly backwards compatible change for either SW or Hardware... if it was vague before, the clarification will both put more restrictions on HW and limit what SW can assume.

timsifive, 2019-01-01:
I would like to require version to always be readable, but since that's a backwards-incompatible change I don't want to make it now. I'll make a PR to make this unambiguous, and to state that the behavior/position of the dmactive bit won't change in the future.

PR #414 (2019-01-07): \Fversion may be invalid when \Factive=0
Fixes #395.
commit b4b3b5c6 (2019-01-07): \Fversion may be invalid when \Factive=0 (#414)

Fixes #395.
closed 2019-01-07 by timsifive