riscv/riscv-debug-spec #407: NMI is not covered in etrigger

chuanhua, 2018-11-12:
etrigger register is defined based on the exception code in mcause CSR. However, NMI is not defined in mcause CSR. NMI is for hardware error conditions. When it happens, it is worthwhile to debug/diagnose the problem. Should it be added to the etrigger definition?

Propose to either add a bit in tdata1 for NMI or use tdata2(XLEN-1) for NMI trigger setup.

timsifive, 2018-11-12:
I'm conflicted whether to put an nmi bit in etrigger or itrigger. itrigger sounds more natural because it deals with interrupts, and nmi stands for non-maskable interrupt. On the other hand an NMI is define as used only for hardware error conditions, which are more of an exception. Thoughts?

chuanhua, 2018-11-14:
I prefer etrigger since NMI is an asynchronous exception that cannot be masked/disabled like a regular interrupt. Also, the current itrigger mechanism only works for the mie/mip CSR interrupt mechanism (called CLINT by SiFive) in the privileged spec. And the new CLIC proposal in the following link may need a new itrigger scheme that can generate a trigger for upto 1024 interrupts. So I think that putting NMI to etrigger will be more stable and not be affected by future privileged spec updates/enhancements.

[CLIC proposal](https://github.com/sifive/clic-spec/blob/master/clic.adoc)

PR #408 (2019-01-09): Add nmi bit to etrigger.
Fixes #407.
commit 1ea1a9b1 (2019-01-09): Add nmi bit to etrigger. (#408)

* Add nmi bit to etrigger.

Fixes #407.

* nmi ignores m/s/u.

The rationale is that nobody can think of a reason that somebody might
want to only trigger on NMI in a specific mode, and by ignoring those
bits it's possible to use a single trigger to trigger on e.g. illegal
instructions in U mode, but on all NMIs.
closed 2019-01-09 by timsifive