riscv/riscv-debug-spec #410: Clarifications are needed for the fields of textra CSR

chuanhua, 2018-11-23:
Two clarifications are needed for the fields of **textra** CSR.

1. textra.**svalue** and textra.**sselect** can be written in **Debug/M/S** modes; textra.**mvalue** and textra.**mselect** can be written in **Debug/M** modes.
2. textra.mvalue and mcontex comparison, when turned on, matches in **M/S/U** modes. textra.svalue and scontext comparison, when turned on, matches in **S/U** modes.

If my proposed clarifications are wrong, please let me know. Thanks.

Chuanhua

ernie-sifive, 2018-12-12:
1. The textra register is at 0x7A3, which is in the range defined by the RISC-V Privileged Architecture as Read/write Machine CSR's.  This register is not writable in S mode.
2. I believe the intent is to check mvalue=mcontext and svalue=scontext regardless of what mode the core is in.

timsifive, 2018-12-12:
The intro in the Trigger Registers section also says:
> The trigger registers are only accessible in machine and Debug Mode to prevent untrusted user code from causing entry into Debug Mode without the OS’s permission.

Agree that the intent is to compare value to context (when enabled) regardless of mode. This can hopefully be used to set a trigger that only applies to a specific process (svalue) running in a specific hypervisor (mvalue).

chuanhua, 2019-05-16:
Another clarification is needed. Should we clarify the presence of textra.sselect and textra.svalue fields with the following sentence?

When S-mode is not implemented, the fields of textra.sselect and textra.svalue will not be implemented as well.

closed 2019-06-26 by timsifive

chuanhua, 2019-07-11:
Based on your previous comments, the scontext CSR in address 0x7AA should not be written by S-mode as well. However, the description says that it can be written by S-mode. Should it be moved to a S-mode CSR address to match the description?

timsifive, 2019-07-15:
Yes, but that would not be a backwards compatible change. I'm not sure what the right solution is here. For now I'll explicitly note that the register address would usually apply it is not writable in S-mode, even though it is.

PR #469 (2019-06-26): Without S-mode, textra.svalue and .sselect should be 0
Fixes #410.
commit 624a6b84 (2019-06-26): Without S-mode, textra.svalue and .sselect should be 0 (#469)

Fixes #410.
PR #474 (2019-07-16): Mention the scontext reg number isn't conventional
It really should be in the S-mode section of CSRs, but that would be an
incompatible change. Fixes #410.
commit 7d126a9d (2019-07-16): Mention the scontext reg number isn't conventional (#474)

It really should be in the S-mode section of CSRs, but that would be an
incompatible change. Fixes #410.