riscv/riscv-debug-spec #415: do triggers match the base address, or any accessed address?

timsifive, 2019-01-08:
Joe Rahmeh reports:

> We have a disagreement within our group over the implementation of sizelo=0, select=0, match=0 and load=1 or store=1 in the match control register (section 5.2.9).  We would appreciate it if you would provide clarification.
> 
> One interpretation of the spec is that the trigger will match when the load/store address matches exactly the value in tdata2.
> 
> Another interpretation is that the trigger will match when the address in tdata2 matches the address of any of the bytes "covered" by the load/store instruction. For example, if the instruction is a "sh" with a data address of a, then the covered bytes are at addresses a and a+1 and the trigger will match if tdata2 equals a or if it equals a+1.
> 
> Which interpretation is correct?
>
> As to our preference, we are still divided.  We would like to implement "cover match" because it is more convenient for debuggers. It would be great if this can be made optional through a bit in mcontrol.  If the bit is zero, then the hardware will perform an exact match.

ernie-sifive, 2019-01-09:
My opinion is that when sizelo=0, any memory access that includes the trigger address should match (your "cover match" interpretation).  If trigger address=4, then a word access to 2 should match, while a byte access to 5 would not match.

timsifive, 2019-01-29:
Consensus at the last meeting was: "Triggers should fire when there’s overlap between the trigger “range” and an access." Which is the same thing that @ernie-sifive said above.

PR #421 (2019-01-31): Address triggers may fire on any accessed address.
Fixes #415.

This change is compatible because the new behavior is optional.
commit 66c31171 (2019-01-31): Address triggers may fire on any accessed address. (#421)

Fixes #415.

This change is compatible because the new behavior is optional.
closed 2019-01-31 by timsifive