riscv/riscv-debug-spec #422: "tselect" and "tdata1" always present when trigger module is not implemented?

chuanhua, 2019-01-31:
From the Trigger Module description (chap. 5), it seems to imply that the "tselect" and ("tdata1" or "tinfo") has to be always present in the CSR space, tselect as R/W and tdata1/tinfo as RO, even when Trigger Module is not implemented.

Is this true? If this is true, then some more explicit description of this fact should be added to clarify it.

Thanks.

Chuanhua Chang
Andes Technology

timsifive, 2019-01-31:
That is true. In systems that don't have any triggers, tselect, tdata1, and tinfo can always return 0 when read, which should be very cheap to implement.

chuanhua, 2019-02-01:
Thank you for the clarification. But, the spec says that tinfo needs to return 1, right?

chuanhua, 2019-02-13:
In the trigger module chapter, it has the following paragraph:
-----
A hart can be compliant with this speciﬁcation without implementing any trigger functionality at
all, but if it is implemented then it must conform to this section.
-----
Then the following paragraphs describes the procedure to find out how many triggers are supported. The quoted paragraph let me think that the procedure is for a design that implements the trigger module. And for a design that does not implement the trigger module, it does not need to conform to the procedure for finding out how many triggers are implemented. And it seems that "tselect" and "tdata1" CSRs do not need to always be present. 

So I think a more explicit clarification is needed.

timsifive, 2019-02-13:
Thinking about this more, if there are no triggers I don't really see a problem with allowing accesses to tselect/tdata* to result in an illegal instruction exception. I agree that the old spec was ambiguous here, but good software should already handle such errors gracefully.

Is this what you had in mind in the first place?

mwachs5, 2019-02-13:
I don't think we can mandate the existence of any of the trigger registers as the priv spec is fairly clear about what registers return 0 vs cause exception if not impelmented. 

I agree there isn't a problem with allowing accesses to trigger registers to result in illegal instruction exception if they are not present.

chuanhua, 2019-02-14:
Either way is fine with me. But if an illegal instruction exception is allowed, could you add the following sentence after the previously quoted paragraph to make this more explicit?

"If the trigger module is not implemented, accessing to any of the trigger module CSRs may/will generate an illegal instruction exception."

closed 2019-03-12 by timsifive