riscv/riscv-debug-spec #427: Should program buffer method and abstract memory access get same result ?

RolfyYu, 2019-02-15:
for example:
addr | 0x0 | 0x1 | 0x2 | 0x3 
---- | --- | --- | --- | --- |
val | AA | BB | CC | DD |
read address 0x1, size is 1

The program buffer method is based on CPU execution
so the result in register is 0x00_00_00_BB (assume using LBU).
How about abstract memory access?
Should data0 be 0x00_00_00_BB or 0x00_00_BB_00 ?

BTW, if program buffer uses LB, the read result will be 0xFF_FF_FF_BB.
Is it also a valid result ?

timsifive, 2019-02-15:
When reading 1 byte using abstract memory access, the result is placed in bits 7:0 of arg0.

The result of RISC-V functions is defined in the ISA spec, and they behave exactly the same in Debug Mode (with minor exceptions documented in Section 4.1 of the Debug Spec).

RolfyYu, 2019-02-18:
Thanks for the clarification.

closed 2019-02-18 by RolfyYu