riscv/riscv-debug-spec #436: JTAG DTM comments

wrightgm, 2019-04-05:
Thanks for all your work on this!

Comments on Version 0.14.0-DRAFT (PDF dated 2/21), JTAG DTM:
* (Clarification) 6.1.2 - "If IR has more than 5 bits, encodings in table 6.1 should be extended with 0s in the msbs." - 1149.1 requires BYPASS to be available with an all-1s encoding.  Is it the intent here that BYPASS be available with both 0-extension and 1-extension?  e.g. if I extend from 5 to 9 bits, must BYPASS be present at 0x000, 0x01f, and 0x1ff ?

* (Functionality) What is the protocol for a single read?
step 1: scan { address, dontcare, op=2'b01 } into dmi
UpdateDR initiates the read to the DM
Now a second scan is required to CaptureDR the read value, and possibly poll on its completion. The question is, what is the op for the second scan? 
According to the table in 6.1.5, op==2'b00 (nop) is defined as never resulting in a busy or error response, and renders addr/data as subsequently undefined.  However, we can't poll with read or write unless we want to initiate a second DM transaction.
I think nop is intended in this case, in which case it might be good to clarify the table with something like:
write 0: Ignore data and address (nop)
Don't send anything over the DMI during Update-DR. If a previous operation is still in progress, the op sticky state will be set.  If the op sticky state is false, Capture-DR returns the address and data of the last successfully completed operation, and a subsequent Capture-DR will return undefined values for address and data.  If the op sticky state is true, Capture-DR returns undefined values for address and data, and the in-progress operation (if any) remains in progress.  A nop will never be in progress during a subsequent DMI operation, and cannot fail.

and possibly also for read value 3 in the table:
The ~~data~~operation scanned into dmi in this access will be ignored.

Appendix B.1, first paragraph, might read:
To read an arbitrary Debug Module register, select dmi, and scan in a value with op set to 1, and address set to the desired register address.  In Update-DR the operation will start.  The result will be captured into data in the Capture-DR of the next dmi operation, which can be a nop, read, or write.  If the read didn't complete in time, that next dmi operation will return 3 in its op field, and the value in data must be ignored.  The busy condition ... [etc]

timsifive, 2019-04-09:
> (Functionality) What is the protocol for a single read?

The way I think of this is that when you scan in a nop while the an operation is in progress, then that nop just gets discarded. The effect that addr/data become undefined doesn't happen because the nop didn't execute. Does that make sense? How can we clarify that?

wrightgm, 2019-04-11:
I will have a go at this section and send you a pull-request when my Foundation membership comes through.

timsifive, 2020-03-26:
Closing due to inactivity.

closed 2020-03-26 by timsifive