riscv/riscv-debug-spec #441: Assumptions on type field of tdata1?

neelgala, 2019-04-22:
Have a couple of questions on the type field of tdata1.

1. Why do we need type=0 to indicate that the trigger doesn't exist? We can get that info from tselect and tinfo itself? Isn't this redundant?

2. From what I am seeing implemented in riscv-openocd ([here](https://github.com/riscv/riscv-openocd/blob/15f2d35bc6413d89453c64eff39a80dfc52243f5/src/target/riscv/riscv.c#L470-L484)) it looks like the type is supposed to be a read-only field and cannot change at runtime. Is that true? If not, I am happy to fix the logic in riscv-openocd which assumes this.

3. Is it mandatory to imlpement all the encodings of the type field (0, 2, 3, 4, and 5)? If not, there should be discovery routine as well to identify which types are supported on an available trigger.

4. Since we are already at 0.14 version does it still make sense to have the SiFive legacy encoding of 1.

timsifive, 2019-04-22:
On Mon, Apr 22, 2019 at 3:38 AM Neel Gala <notifications@github.com> wrote:

> Why do we need type=0 to indicate that the trigger doesn't exist? We can
> get that info from tselect and tinfo itself? Isn't this redundant?
>
It's allowable to make tselect e.g. 3 bits wide, even if there are fewer
than 8 triggers. The idea was that this might be a simpler implementation
option than putting extra logic on tselect accesses to make sure e.g. 7
cannot be written there. In that case trigger 7 can be selected, and when
read back its type will show to be 0.

From what I am seeing implemented in riscv-openocd (here
> <https://github.com/riscv/riscv-openocd/blob/15f2d35bc6413d89453c64eff39a80dfc52243f5/src/target/riscv/riscv.c#L470-L484>)
> it looks like the type is supposed to be a read-only field and cannot
> change at runtime. Is that true? If not, I am happy to fix the logic in
> riscv-openocd which assumes this.
>

Type may not be read-only, but OpenOCD doesn't support writing the type.

Is it mandatory to imlpement all the encodings of the type field (0, 2, 3,
> 4, and 5)? If not, there should be discovery routine as well to identify
> which types are supported on an available trigger.
>

No. Everything about triggers is optional. It's assumed the debugger will
attempt to set whatever trigger it wants, and then discover if it happened
or not. If so it can try something else or report an error to the user.

For types specifically, the tinfo register can be used to discover which
types are supported by a trigger.

Since we are already at 0.14 version does it still make sense to have the
> SiFive legacy encoding of 1.
>

Yes, because there is hardware (specifically HiFive1) that implements it.
When we run out of types, probably enough time has gone by that we can
reuse this encoding.

Tim

>

neelgala, 2019-04-23:
> It's allowable to make tselect e.g. 3 bits wide, even if there are fewer than 8 triggers. The idea was that this might be a simpler implementation option than putting extra logic on tselect accesses [...]

It would be better for `tinfo` to return all zeros when trigger 7 is read, instead of reserving `0` in `type` to achieve the same. Plus with the current draft, the LSB of `tinfo` for all valid triggers will always be hardwired to `0`. Looks like we are loosing out on bits to accommodate this feature.

> Type may not be read-only, but OpenOCD doesn't support writing the type. 

From what I can tell, opencd currently supports only MControl type? I have a fix which will enable writing the type field. It isn't that difficult. But what is required is reading tinfo to identify if the mcontrol type is supported on the selected trigger. I could do a small PR for that if you okay with it? 

> No. Everything about triggers is optional. It's assumed the debugger will attempt to set whatever trigger it wants, and then discover if it happened or not. If so it can try something else or report [...]

Thanks for clarifying. Its just that a few fields explicitly mention they are optional, I thought it would make sense to either put the optional everywhere or nowhere. A mix of it is confusing.

timsifive, 2019-04-23:
On Mon, Apr 22, 2019 at 9:54 PM Neel Gala <notifications@github.com> wrote:

> It would be better for tinfo to return all zeros when trigger 7 is read,
> instead of reserving 0 in type to achieve the same. Plus with the current
> draft, the LSB of tinfo for all valid triggers will always be hardwired
> to 0. Looks like we are loosing out on bits to accommodate this feature.
>
> 0.14 is a draft, but we committed to keeping it backwards and forwards
compatible with 0.13 which is ratified. We can't change the encoding of the
type field and remain compatible.

> From what I can tell, opencd currently supports only MControl type? I have
> a fix which will enable writing the type field. It isn't that difficult.
> But what is required is reading tinfo to identify if the mcontrol type is
> supported on the selected trigger. I could do a small PR for that if you
> okay with it?
>
> What would a user do to set triggers that don't use the mcontrol type?
AFAIK what kind of hardware breakpoints can be communicated between gdb and
OpenOCD is very limited.

> No. Everything about triggers is optional. It's assumed the debugger will
> attempt to set whatever trigger it wants, and then discover if it happened
> or not. If so it can try something else or report [...]
I'll try to fix this in #444.

Tim

>

timsifive, 2019-04-25:
@neelgala can you confirm that #444 makes the spec more clear?

timsifive, 2020-03-26:
Closed due to inactivity.

closed 2020-03-26 by timsifive