riscv/riscv-debug-spec #445: Matching on addresses other than base address

pdonahue-ventana, 2019-04-24:
In section 5.2.9 of the 0.14.0 draft, a _select_ value of 0 is defined to
> Perform a match on the virtual base address of the access. (E.g. on a 32-bit read from 0x4000, the base address is 0x4000.)

This says that the comparison is done on the base address, not on any other address.  In the same section, a _match_ value of 0 is (after https://github.com/riscv/riscv-debug-spec/pull/421) defined as
> Matches when the value equals tdata2. Additionally, if select =0 then it is recommended that the trigger also matches if any of the accessed addresses equal tdata2. (E.g. on a 32-bit read from 0x4000, the following addresses are accessed: 0x4000, 0x4001, 0x4002, and 0x4003.)

This explicitly says that select=0 can match on addresses besides the base address.

The definition of select=0 should be updated to be consistent with the recommendation in the _match_ text.  For instance, simply saying this would leave further explanation to the _match_ description:
> 0: Perform a match on the virtual address of the access.
> 1: Perform a match on the data value loaded or stored, or the instruction executed.

I'm also unsure if the recommendation applies to more than just match=0.  It is clear that the recommendation is that this case should match:
select=0
match=0
tdata2=0x4002
32-bit read with a base address of 0x4000

But if we have this case (which is the same except that some masking is applied), is it recommended to or even allowed to match?
select=0
match=4
tdata2=0xF0004002 and XLEN=32
32-bit read with a base address of 0x4000

Ideally, the matching behavior would be described with pseudocode to remove any ambiguity.

timsifive, 2020-03-26:
Closing because a resulting change merged. Please reopen if that did not address the problem.

closed 2020-03-26 by timsifive

pdonahue-ventana, 2020-03-26:
You made me go back and figure out what I was thinking 11 months ago. :-)

The latest spec looks fine with respect to this.

timsifive, 2020-03-27:
Thanks for checking.