riscv/riscv-debug-spec #448: Name of the document

pdonahue-ventana, 2019-04-24:
The draft version 0.14.0 of the debug spec is titled "RISC-V External Debug Support".

Since the functionality described in the document provides for debugging via software with no external debugger (via action=0) and there is no competing RISC-V proposal for non-external debug, it seems that the name of the document should be changed from "RISC-V External Debug Support" to the more general "RISC-V Debug Support".

timsifive, 2019-04-25:
"Debug" could be interpreted to be a lot broader than "External Debug," but I don't have a strong opinion.
If you can get some other people to agree that this is a good change I'm happy to change it.

pdonahue-ventana, 2019-04-25:
> "Debug" could be interpreted to be a lot broader than "External Debug,"

Exactly.  Because the spec addresses a broader set of issues than external debug, the title is misleading.  If it described only OpenOCD-style debugging then the current title is fitting.  Since it also addresses gdb-style debugging, a more generic name is appropriate.

I'll ask on tech-debug.

pashenden, 2019-04-25:
Works for me.

PR #581 (2020-11-03): RISC-V External Debug Support -> RISC-V Debug Support
Fixes #448
commit b860d530 (2020-11-03): RISC-V External Debug Support -> RISC-V Debug Support (#581)

Fixes #448
closed 2020-11-03 by timsifive