riscv/riscv-debug-spec #453: ebreaks and ebreaku

pdonahue-ventana, 2019-05-17:
What happens to ebreaks and ebreaku in dcsr if S and/or U modes aren't supported?  Are those R/W bits still required to contain storage elements but they simply don't do anything?  Do those bits become hardwired?

Should these bits be declared as WARL and, if so, what is the legal/hardwired value for an unsupported mode?  (Zero seems natural and matches dcsr[14] for the unsupported mode=2.)

timsifive, 2019-05-20:
I'm inclined to add: "This bit is wired to 0 if the hart does not support S mode." and similar for ebreaku.

How does that sound?

pdonahue-ventana, 2019-05-20:
I think that hardwiring to 0 sounds good for unsupported modes.

Shouldn't those bits also be defined as WARL?  I think that WARL was created exactly for this type of thing.

PR #458 (2019-05-28): ebreaks/ebreaku changes
Fixed #453
Defined ebreaks and ebreaku as WARL fields that are 0 if the associated mode is not supported by the hart.
commit 633ee13c (2019-05-28): Fixed #453 (#458)

Defined ebreaks and ebreaku as WARL fields that are 0 if the associated mode
is not supported by the hart.
closed 2019-05-28 by timsifive