riscv/riscv-debug-spec #455: mcontrol.size

pdonahue-ventana, 2019-05-18:
In 0.14.0-DRAFT, the sizehi field of mcontrol is defined as a R/W field and described as "This field only exists if XLEN is greater than 32.  In that case it extends size. If it does not exist then hardware operates as if the field contains 0."

I have a few questions:

1. If sizehi does not exist (because XLEN=32), does that mean that this field is no longer R/W?  If so, could it just be WARL instead of not existing?  If not, does an RV32 implementation need to implement two state bits for this useless field simply to support R/W?
2. To further complicate matters: Section 5.2 says "In this section XLEN means MXLEN when in M-mode..."  If software on an RV64 implementation changes the MXL field in misa from 2 to 1 then XLEN goes from 64 to 32.  By the current definition, that means that sizehi previously existed but no longer exists.  What are the implications of this?
3. If an implementation has XLEN=32 but supports execution of instructions that are 48-bits or more, shouldn't sizehi be required or at least allowed?

I think that the sizehi description should only say:
"This field contains the 2 high bits of size.  See sizelo for more details."

Then either sizelo/sizehi could be WARL or the sizelo description could add: "Implementations with XLEN<=64 and that do not support instructions of length > 96 may hardwire size[3]=0.  Implementations with XLEN=32 and that do not support instructions of length > 32 may hardwire size[2]=0."

timsifive, 2019-05-29:
Does #460 address your concerns?

pdonahue-ventana, 2019-05-29:
Looks good.  Thanks.

PR #460 (2019-06-07): Clarify mcontrol.size.
My attempt to fix #455.
commit b06eb70d (2019-06-07): Clarify mcontrol.size. (#460)

* Clarify mcontrol.size.

My attempt to fix #455.

* Fix cut and paste error.
closed 2019-06-07 by timsifive