riscv/riscv-debug-spec #462: exceptions in quick access

timsifive, 2019-06-04:
From Marc Gauthier:
The sequence shows that if any program buffer instruction takes an exception, the hart resumes execution immediately.  This is generally fatal to the program being debugged, as it will corrupt its state.  Most program buffer sequences save some registers, do an operation, and restore the registers (very little can be done otherwise).  If the operation hits an exception and the hart resumes immediately, registers are not restored, resulting in program state corruption.  This effectively means the operation must not take an exception.  But if it must not take an exception anyway – leaving the hart halted is much better than corrupting its state, no matter how bad halting may be.  Also note that not being able to take an exception is particularly limiting:  it means the debugger cannot offer the user arbitrary access to memory, such as to browse virtual address spaces etc, because the debugger has no way to know whether an access at a particular address will succeed.  Attempting it anyway and giving the responsibility to the user is fraught with peril:  failure is silently fatal (the worst kind of bug, especially when introduced by a debugger).

At least if the hart stays halted, the debugger can clean things up and resume correct execution.

A clean and non-intrusive solution is an optional proper trap mechanism inside the program buffer:  a way to designate where in the program buffer to jump to if an exception occurs.  Or if the implementation supports branches within the program buffer, start execution at the 2nd program buffer word, and jump to the 1st program buffer word on exception; the instruction at the first word can branch to the  appropriate restore sequence.  And either way, if a second exception occurs, leave the hart halted.

pdonahue-ventana, 2019-06-04:
I agree that remaining halted seems sensible.

The external debugger is going to notice that there was an exception and infer that the quick access failed to give it the information it wanted.  It's going to have to deal with that situation in some way and restoring a few registers is a minor extra burden.

bruceable, 2019-06-04:
Are you saying there is a DM flag that indicates an exception has been
raised while in debug mode?

I believe that in debug mode all interrupts and exceptions are masked, thus
debug mode maintains control of the cpu and can only be relinquished when
that code returns to normal execution (M, S, U).

pdonahue-ventana, 2019-06-04:
> Are you saying there is a DM flag that indicates an exception has been raised while in debug mode? I believe that in debug mode all interrupts and exceptions are masked, thus debug mode maintains control of the cpu and can only be relinquished when that code returns to normal execution (M, S, U).

Interrupts are masked per bullet 2 in section 4.1.  Exceptions still happen, though they don't update any registers per bullet 3.  Then cmderr is set to 3: "An exception occurred while executing the command (e.g. while executing the Program Buffer)."

timsifive, 2020-11-03:
The simplest solution is to leave the hart halted as Marc suggested. I'll make a PR for that. AFAIK nobody has yet implemented quick access, so I don't think it warrants adding more complexity.

PR #585 (2020-11-07): Change quick access exceptions to halt the target.
Fixes #462.
commit c89895ca (2020-11-07): Change quick access exceptions to halt the target. (#585)

Fixes #462.
closed 2020-11-07 by timsifive