riscv/riscv-debug-spec #468: [debug] Abstract Command : Access Registe GPR Mandatory ?

command-paul, 2019-06-17:
Did not get any feedback on the debug google group so raising an issue here.

I was drawing up an implementation of a debug spec compliant debug module for integration with an out of order core and was looking for something that was minimally invasive in a structural sense.

The debug spec mandates the presence of the Access Register : Abstract Command 
at :(spec 14.0-Draft , Chapter 3 , page 7 ). 

```
2. Allow any individual hart to be halted and resumed. (Required) 
3. Provide status on which harts are halted. (Required) 
4. Provide abstract read and write access to a halted hart’s GPRs. (Required)
```
Working in the space of my structural constraints , i.e. no abstract commands. I find that it the spec can comfortably accommodate debugging any riscv target implementing a simple combination of ::

1] Memory Access Through System Bus Access.
2] GPR/FPR/CSR by using the program buffer / quick access.

while respecting the present specifications of the Program Buffer  and  System bus access .

Are there any fundamental limitations to the above I may be missing ?

This configuration would need a few updates to riscv-openocd but like its present elegant implementation would not need any explicit run time target specific config as all features are still discoverable.

Can the Hard Requirement for Abstract GPR Access be softened or be redefined to accommodate such a possible valid alternative implementation?

Could this be Version 3 ?

timsifive, 2020-02-10:
What you've described actually gets us closer back to 0.11. :-) (Except that didn't allow for abstract commands at all.)

If you want minimally intrusive, the best way is probably to use a debug ROM type solution like rocket chip does. Then you can extend it a bit and convert abstract register access commands to jumping to the right place in ROM, or use some hybrid where part of the "ROM" is really changed depending on which abstract command is being performed. There is nothing in the spec that requires abstract access to be implemented with additional ports on your register file.

If you really want to push for this change, your best bet is to start a discussion on the tech-debug mailing list. Relatively few people look at the issues here.

timsifive, 2020-10-30:
Feel free to reopen if this still needs to be addressed.

closed 2020-10-30 by timsifive