riscv/riscv-debug-spec #477: Minimal RISC-V Debug Spec

CLappin, 2019-07-24:
Hi all,

I was having a read of the latest spec and it mentions a minimal debug spec document which uses the abstract commands. Has this document been published yet?

Thanks, 
Ciaran

timsifive, 2019-07-26:
I don't think there is such a document. Where do you see it referenced?

CLappin, 2019-07-26:
Hi Tim, 

In Chapter 3 Debug Module (pg 7) of the draft 0.14 spec: ( Second paragraph point 3c ).

Implement abstract access to at least all GPRs, dcsr, and dpc, and advertise the implementation
as conforming to the “Minimal RISC-V Debug Specification 0.14.0-DRAFT”, instead of the “RISC-V Debug Specification 0.14.0-DRAFT”.

I'm very interested in this minimal spec for a core I'm working on at the moment. 

Thanks, 
Ciaran

timsifive, 2019-07-26:
This isn't a reference to a separate document, but it outlines a minimum implementation that may not provide CSR/memory access but might still be useful. Phrased otherwise, a minimum implementation must implement all the required features listed at the beginning of Chapter 3, as well as abstract access to DPC and DCSR.

CLappin, 2019-07-31:
Thanks Tim.

closed 2019-07-31 by CLappin