riscv/riscv-debug-spec #480: havereset

pdonahue-ventana, 2019-08-06:
> Individual harts (or several at once) can be reset by selecting them, setting and then clearing
hartreset.

> When harts have been reset, they must set a sticky havereset state bit.

Does havereset get set on the positive edge of hartreset or on the negative edge?

timsifive, 2019-08-13:
havereset gets set when a hart has been reset. Depending on the reset mechanism, when that happens in relation to hartreset is implementation-dependent. It may be on either edge, or at a different time altogether.

pdonahue-ventana, 2019-08-13:
Assuming nonexistent=0, are there any constraints on the combination of havereset, unavail, running, and halted observed by the external debugger?

timsifive, 2019-08-14:
From Hart DM States:
"Every hart that can be selected is in exactly one of the following four DM states: non-existent, unavailable, running, or halted. Which state the selected harts are in is reflected by allnonexistent, anynonexistent, allunavail, anyunavail, allrunning, anyrunning, allhalted, and anyhalted."

pdonahue-ventana, 2019-08-14:
Yes, but my question was about havereset with respect to these DM states.

Here's an example.  You said that havereset can assert at some arbitrary implementation-dependent time ("a different time altogether").  Let's say that the implementation asserts havereset 1000 cycles after reset deasserts.  Can an external debugger observe running=1 and havereset=0 during that window?

timsifive, 2019-08-15:
Yes, but I would interpret that to mean that the hart has not yet reset, or is in the process of resetting.

If I had to be more precise about havereset, I would say:
havereset is asserted after reset is complete, and before the first instruction out of reset is executed. (Or fetched, even?)

pdonahue-ventana, 2019-08-17:
> Yes, but I would interpret that to mean that the hart has not yet reset, or is in the process of resetting.

That seems to misrepresent the situation since running=1, havereset=0 means either that everything is fine or that there was a very recent reset.  It seems better to me to:
- require unavail=1 when reset asserts (rather than the current "may" language)
- transition to unavail=0 no earlier than when havereset asserts

> If I had to be more precise about havereset, I would say:
> havereset is asserted after reset is complete, and before the first instruction out of reset is executed. (Or fetched, even?)

I would say that:
- havereset must assert no earlier than the assertion of unavail
- havereset must assert no later than when running or halted assert (same as the second bullet above)
- running must assert before the first post-reset instruction is fetched

Whether your idea or mine, can any requirements in this area be specified so the debugger can rely on them?

timsifive, 2019-08-19:
Good point about unavail. I don't see where you see "may" used for that. I thought I just read through all the relevant reset text, but it sounds like I missed something. Can you point me at it?

I did find: "The system may take an arbitrarily long time to come out of reset, as reported by allunavail, anyunavail." That implies that harts are unavail when resetting.

Building on that and clarifying, a reset looks like this:
1. debugger asserts some reset
2. unavail becomes set
3. debugger deasserts that reset
4. time passes
5. havereset=1, unavail=0. The hart may either be running or halted depending on haltreq and resethaltreq.
6. debugger clears havereset by writing ackhavereset=1

I could make an argument for allowing unavail to stay 0 until step 4. In that case an external reset controller could be signaled by the debugger to make the reset happen, and unavail directly reflects whether the hart is held in reset. This would allow debugging to continue if the reset controller isn't doing its job for some reason.

pdonahue-ventana, 2019-08-19:
I'm looking in the section on hart state (which defines unavail, running,
halted).

> Harts may be unavailable for a variety of reasons including being
reset ...

This seems to allow for reset => unavailable but doesn't seem to impose a
requirement.

> Which states a hart that is reset goes through is implementation
dependent. Harts may be unavailable while reset is asserted, and some time
after reset is deasserted. They might transition to running for some time
after reset is deasserted. Finally they end up either running or
halted, depending on haltreq and resethaltreq

This language seems very clear.  Which states are implementation dependent,
not simply the timing of a required sequence of states.  It also states
that a hart "may" be unavailable while reset is asserted (echoing the "may"
language of the first quote above).  The only requirements are the final
post-reset state in the last sentence and presumably the overriding
requirement that a hart can't ever temporarily become nonexistent.

Thanks,

-Paul

timsifive, 2019-09-17:
That sounds reasonable. The reset now looks like this:
1. Debugger asserts some reset.
2. Debugger deasserts that reset.
3. Time passes.
4. Actual reset begins. Harts move to either unavail or running.
5. Time passes.
6. havereset=1, unavail=0. The hart may either be running or halted depending on haltreq and resethaltreq.
7. debugger clears havereset by writing ackhavereset=1

Step 4 might occur as early as immediately after step 1.

Did I get that right?

pdonahue-ventana, 2019-09-17:
Yes, that sounds good.

PR #494 (2019-10-02): Rewrite/clarify DM Reset Control
Hopefully fixes #480.
commit d933becb (2019-10-02): Rewrite/clarify DM Reset Control (#494)

* Rewrite/clarify DM Reset Control

Hopefully fixes #480.

* Clarifications from review.

* Another clarification.

* There's no way to tell when reset has started.

I don't think that's very important. If it is, we'll probably have to
add a status bit that indicates a hart is held in reset.
closed 2019-10-02 by timsifive