riscv/riscv-debug-spec #492: Resuming the CPU when halted due to trigger

JanMatCodasip, 2019-09-16:
Let's assume a processor halts due to a type-2 trigger that is set to fire _before_ execution of the instruction (i.e. mcontrol.timing=0). Is my understanding correct that resuming the processor (dmcontrol.resumereq <-- 1) without disabling the trigger shall cause the trigger being re-activated? In other words, no progress in the program is made until such trigger is removed?

Thank you for clarification,
Jan

timsifive, 2019-09-16:
That is right. If that trigger fires e.g. on the instruction at 0x1000, and dpc=0x1000 when resuming from Debug Mode, then the trigger will fire immediately. The instruction will not be executed.

JanMatCodasip, 2019-09-17:
Thank you for checking my interpretation of the spec, appreciated.

closed 2019-09-17 by JanMatCodasip