riscv/riscv-debug-spec #495: Pulsing "dmactive" low vs. hart state?

JanMatCodasip, 2019-09-27:
Question regarding the "dmactive" bit (in "dmcontrol" reg. of the RISC-V debug module) - 

Is my understanding of the spec correct that pulsing the "dmactive" bit low (thereby resetting the debug module) shall not affect the state of harts? I.e. halted harts will remain halted, running harts will remain running?

Thank you,
Jan

timsifive, 2019-09-27:
Right. Section 3.2 says: "The halt state of harts should be maintained across system reset provided that dmactive is 1, although trigger CSRs may be cleared."

JanMatCodasip, 2019-09-30:
> Section 3.2 says: "The halt state of harts should be maintained across system reset provided that dmactive is 1, although trigger CSRs may be cleared."

Thanks for replying. Actually, although close, I believe the quoted sentence from the spec does not describe the situation I'm trying to get confirmation on.

In my situation, no system reset takes place. Only dmactive is set to 0 for a moment, then set back to 1, thereby resetting just the debug module. In that case, is it obligatory that hart states must not change (harts must remain running/halted/..., same as prior deasserting dmactive)? Or, would it be permissible to e.g. resume halted harts when the debug module undergoes its reset?

Thanks.

timsifive, 2019-10-01:
I can imagine two different system designs:
1. The DM emits some kind of signal, and when that signal is high the hart is halted. Resetting the DM would cause the harts to resume.
2. The DM sends separate signals (complex operations, whatever) which cause a hart to halt/resume. The hart keeps track of its own halted state. Resetting the DM would not affect the hart's state.

These both seem legit, so I think either behavior ought to be allowed. I'll propose a clarification to that effect, and see how it goes over.

timsifive, 2019-10-01:
Although the system reset line ("The halt state of harts should be maintained across system reset provided that dmactive is 1, although trigger CSRs may be cleared.") implies #1 above. So I guess you're definitely in the clear letting your harts run when the DM is reset.

JanMatCodasip, 2019-10-02:
Thanks for your point of view. That fully answers my question, so I'll close this thread.

I agree, adding a sentence to the spec that would clarify the matter would be, IMO, helpful.

closed 2019-10-02 by JanMatCodasip