riscv/riscv-debug-spec #509: Is it more appropriate to move the NMI field from etrigger to itrigger?

Melflyu, 2020-01-13:
I had reported a question about NMI clarification earlier. Krste Asanović professor pointed out that NMI is considered as an interrupt. Please refer the web-page below:
https://github.com/riscv/riscv-isa-manual/issues/473

In the ISA specification, the NMI stands for non-maskable interrupt and use to report the machine error. The NMI seems more like an interrupt from this perspective even it is un-maskable. And this is also the same as the professor's reply.

According to #407 . The NMI is considered as an exception in RISC-V debug specification. This is why the debug specification adds a field to exception trigger(etrigger) for NMI. And this filed regardless of the m, s, and u mode.

Combining the results of the Krste Asanović professor's reply and #407 , is it more appropriate to move the NMI field from etrigger to itrigger? 

And the NMI filed definitions can refer to current specifications. When this optional bit is set, non-maskable interrupts cause interrupt trigger to fire, regardless of the values of m, s, and u.

timsifive, 2020-01-13:
This was briefly discussed in https://github.com/riscv/riscv-debug-spec/issues/407. If this is just a discussion about naming then I'm inclined to leave things as-is because there may already be some implementations.

Melflyu, 2020-01-14:
I think put NMI in etrigger is not a good idea, because the `Instruction Set Manual` specification has clearly defined interrupt and exception in section 1.4. Please see the description below:
> We use the term exception to refer to an unusual condition occurring at run time associated with **instruction** in the current RISC-V hart. We use the term interrupt to refer to an **external asynchronous event that may cause a RISC-V hart to experience an unexpected transfer of control**.

•	NMI is non-maskable “interrupt”. 
      o	It is not straightforward to define NMI as an exception.
      o	The NMI causes a RISC-V hart to experience an unexpected transfer of control.

•	NMI is an asynchronous event, and asynchronous events can be taken without valid instruction.
      o	If nmi is in etrigger. Logic should test etrigger even if there is no valid instruction.

timsifive, 2020-03-26:
None of this makes a functional difference, though. Where the bit is doesn't matter to the hardware. If we had had your input when defining this feature then the bit may have ended up in the other trigger type. If you still feel this should be changed, I suggest making your case on the tech-debug mailing list. Not that many people follow the issues.

timsifive, 2020-10-30:
Please reopen if this is still important.

closed 2020-10-30 by timsifive