riscv/riscv-debug-spec #510: What the behavior should be when interrupts are enabled during hardware single-stepping?

Melflyu, 2020-01-13:
The debug specification defines the hardware single-step behavior after exception and instruction complete. please see the description below:
> An external debugger can cause a halted hart to execute a single instruction or trap and then re-enter Debug Mode by setting _step_ before setting _resumereq_.
> If executing or fetching that instruction causes an exception, Debug Mode is re-entered immediately after the PC is changed to the exception handler and the appropriate _tval_ and _cause_ registers are updated.

The interrupts are enabled during hardware single-stepping when the both of _stepie_ and _step_ field of _dcsr_ register are set to 0x1. But this behavior is not defined to RISC-V debug specification. This has two possible consequences. 

If debug mode is entered after interrupt. the execution sequence would similar to below:
T0: Single-step resume
T1: Interrupt taken. The single-step trigger is fired.
T2: Enter debug mode
T3: Execute first instruction on the Interrupt trap handler
The _DPC_ would be set to the PC of instruction (T0+2 or 4) after T2 debug mode. This will cause debug mode is entered on the same _DPC_ with hardware single-step _dcause_. 

The CSR mip/sip/uip must be scanned in debug handler under this behavior when two DPC are the same after two entering consecutive debug mode

If debug mode is not entered after interrupt. the execution sequence would similar to below:
T0: Single-step resume
T1: Interrupt taken.
T2: Execute first instruction on the interrupt trap handler. The single-step trigger is fired.
T3: Enter debug mode
The _DPC_ would be set to the PC of instruction (T2+2 or 4) after T3 debug mode. Exception vector could be used to found the taken interrupt.

Which behavior is the more correct behavior when interrupts are enabled during hardware single-stepping?

pdonahue-ventana, 2020-01-13:
I think that the quoted sentence is incorrect because not all exceptions cause traps and the term "exception handler" is not used or defined in the base architecture.  The base architecture says: "We use the term trap to refer to the transfer of control to a trap handler caused by either an exception or an interrupt."

I think that this is probably what was intended:
"If executing or fetching that instruction causes a trap, Debug Mode is re-entered immediately after the PC is changed to the trap handler and the appropriate tval and cause registers are updated."

If so, that would answer the question: Like with exceptions which trap, changing the PC to the trap handler for the interrupt is a step that comes before executing the first instruction of the handler.

timsifive, 2020-01-13:
What should happen is:

timsifive, 2020-02-10:
Please reopen if #511 didn't address this.

closed 2020-02-10 by timsifive