riscv/riscv-debug-spec #519: debugger should be able to request that a hart stay powered on

timsifive, 2020-04-01:
subject:"Abstract command related questions and suggestions"

Robert Chyla:
> Some existing processors provide a logic to prevent (selective) core[s] from going into some deep-sleep/not-clocked/not-powered modes. This is usually done in SoC specific way, but maybe commonly agreed 'RISC-V debug logic' should have dedicated bit for it? This way debugger may be relieved from doing 'magic' in SoC specific way. Some SoC may ignore it, but if someone is 'compliant' debugging of hard (but practical ...) conditions will be pleasant and predictable.

timsifive, 2020-10-28:
Is this intended to prevent halted harts from becoming unavailable, or running ones?

For halted harts, we can simply say that hardware should attempt to keep halted harts available. For running harts, we could add 2 bits to dmcs2: setkeepavailable and clearkeepavailable. They would set/clear the keepavailable status for all currently selected harts. If keepavailable is set, we request that hardware keep the harts available, if possible.

Thoughts, @pdonahue-ventana?

pdonahue-ventana, 2020-10-30:
This looks like a good way to request keepavailable to be set/cleared.

The actual availability is usually managed by something (one of the harts or a management processor or even some hardware state machine) that may take some time to actually see the change to keepavailable.  To avoid the race condition, it would be useful to have an acknowledgement mechanism like on dmactive.

It's a bit more complicated than dmactive since this affects multiple harts.  How about treating setkeepavailable like allkeepavailable on reads and clearkeepavailable is anykeepavailable on reads?  Something like:
> setkeepavailable: Writing 0 has no effect.  Writing 1 requests that the keepavailable bit be set for all selected harts.  On reads, this returns the AND of the actual keepavailable bits for each selected hart.  After writing 1, the debugger must poll until this bit reads as 1, indicating that the setkeepavailable request has been acknowledged for all selected harts and all of them have seen that keepavailable=1.

> clearkeepavailable: Writing 0 has no effect.  Writing 1 requests that the keepavailable bit be cleared for all selected harts.  On reads, this returns the OR of the actual keepavailable bits for each selected hart.  After writing 1, the debugger must poll until this bit reads as 0, indicating that the clearkeepavailable request has been acknowledged for all selected harts and all of them have seen that keepavailable=0.

timsifive, 2020-10-30:
I don't understand why any race might matter. Setting the bit requests that the hardware keeps the hart available. There is no guarantee that the hardware can honor that request, so there's no need to know that some other part of the platform noticed it, is there? I assume the bit itself is in the DM, and can be updated near-instantly (or at least before a change is made to which hart is currently selected).

pdonahue-ventana, 2020-10-30:
I assume that the bit itself logically lives in the power controller (but physically may live anywhere).  Until the power controller is aware that keepavailable was set, it may power off the hart.

It's similar to CDBGPWRUPREQ and CDBGPWRUPACK in Coresight v3.  The debugger requests that a core (or other component) is powered up and then it waits until the power controller acknowledges that it saw the request.  The power controller will not power down that core until the debugger removes the request.

It's basically the same as dmactive which has an acknowledgement mechanism.

timsifive, 2020-11-04:
I've written up something with set/clr all/any bits. How do we handle support?

It seems possible that a single DM is connected to multiple harts, and some of those harts support keepalive and some do not. The debugger needs to know so it doesn't set keepalive and then waits for allkeepalive to go high, because it may not happen. Unless we define the bit as being set when the DM sees it, and then whether something actually happens depends on the rest of the system. But if the bit is in the DM, do we even need the all/any bits? Thoughts?

ernie-sifive, 2020-11-04:
It seems sufficient to me to simply be able to set and clear a keepalive output bit from the DM for each hart it is connected to.  There is no guarantee that the hardware will respond or if it does, how long it will take.  As long as a core is powered down, presumably its unavail bit is set.  This informs software that the hart cannot respond to commands and I don't see the need for an additional status bit per hart.

timsifive, 2020-11-05:
I agree with @ernie-sifive, but @pdonahue-ventana argued (https://github.com/riscv/riscv-debug-spec/issues/519#issuecomment-719846264) that an acknowledgment mechanism is important. I think the hardware cost of making the bit readable (assuming it lives in the DM) is relatively low, and at least that allows a more complex implementation where the bit lives somewhere else in the system. Did I miss anything?

commit 9fea4c58 (2020-11-13): Add keepalive feature. (#592)

* Add keepalive feature.

This allows a debugger to request that a given hart stays available for
debugging (e.g. is not powered down). Fixes #519.

* Remove keepalive status reading.

* writes -> sets

* Make the example more conservative.

Feedback from Robert Chyla.
closed 2020-11-13 by timsifive