riscv/riscv-debug-spec #522: add appendix use case: how to halt at first instruction out of reset

mwachs5, 2020-04-21:
there are lots of nice use cases in the appendix. I think another useful one could be "halting at first instruction out of reset"

dsula, 2023-04-07:
That's exactly what I need. How do you reset and halt at first instructions? I'm not able to find this in the documentation. There's no DMCONTROL.resethaltreq in my implementation, yet the debugger has no problem halting at first instruction. How is that done?

timsifive, 2023-04-07:
Assuming the debugger you're using is OpenOCD, I suggest capturing a log when running it with `-d3` and you'll be able to see exactly what OpenOCD is doing. It might take some time to find the right part in the log, as it will be huge, though. ;)

dsula, 2023-04-10:
Yes, I was able to wade thru the openocd log file and got the sequence. The sequence was as expected. Interestingly however, the sequence does not work when implementing as a custom reset using the segger jlink tools. But thanks for your suggestion.