riscv/riscv-debug-spec #523: Specify a DMI port for Debug Modules

kmeinhar, 2020-05-04:
I am student that did a research project on open-source implementations of the debugging specification.

What I noticed was that a lacking specification of an interface to the Debug Module resulted in multiple different interfaces that are similar but not compatible with each other.

### The Problem:

The specification reads:

> The Debug Module Interface can be a trivial bus with
one master and one slave, or use a more full-featured bus like TileLink or the AMBA Advanced
Peripheral Bus. The details are left to the system designer.

And:

> The DMI uses between 7 and 32 address bits. It supports read and write operations.

The only mention of signals to and from the Debug Module are in the description of JTAG-DTM.
There register `dmi` specifies the three fields: `address`, `data` and `op`.

The inspected open-source implementations took those three fields as an inspiration for an interface to the Debug Module.
This lead to the following interfaces:

|Signal Name    | [PULP](https://github.com/pulp-platform/riscv-dbg)| [SweRV](https://github.com/chipsalliance/Cores-SweRV)| [SCR1](https://github.com/syntacore/scr1)| [Rocket Chip](https://github.com/chipsalliance/rocket-chip)   | [Piccolo](https://github.com/bluespec/Piccolo)  | Piccolo JTAG TAP |
| ------------- |-------------		| -----			|-----		|-----		| ---		| ---		|
|Request Ready 	| dmi\_req\_ready\_o 	| - 			|  - 		| dmi.req.ready | -				|jtagtap.dmi.req\_ready		|
|Request Valid 	| dmi\_req\_valid\_i 	| dmi\_reg\_en 		| dmi\_req	| dmi.req.valid | -				|jtagtap.dmi.req\_valid		|
|Request Address| dmi\_req\_i.addr 	| dmi\_reg\_addr 	| dmi\_addr 	| dmi.req.addr 	| dm\_dmi.read\_addr(addr)	|jtagtap.dmi.req\_addr		|
|Request Data 	| dmi\_req\_i.data 	| dmi\_reg\_wdata 	| dmi\_wdata 	| dmi.req.data 	| dm\_dmi.write(addr, data)	| jtagtap.dmi.req\_data		|
|Request Op	| dmi\_req\_i.op 	| - 			| - 		| dmi.req.op 	| -				| jtagtap.dmi.req\_op		|
|Respond Ready 	| dmi\_resp\_ready\_i 	| - 			| - 		| dmi.resp.ready| -				|jtagtap.dmi.rsp\_ready		|
|Respond valid 	| dmi\_resp\_valid\_o 	| - 			| dmi\_resp 	| dmi.resp.valid| -				|jtagtap.dmi.rsp\_valid		|
|Respond Data 	| dmi\_resp\_o.data 	| dmi\_reg\_rdata 	| dmi\_rdata 	| dmi.resp.data | dm\_dmi.read\_data 		|jtagtap.dmi.rsp\_data		|
|Respond Status | dmi\_resp\_o.resp 	| - 			| - 		| dmi.resp.resp | -				|jtagtap.dmi.rsp\_response		|
|Reset Signal 	| dmi\_rst\_ni 		| - 			| - 		| - 		| -				|-		|
|Write Enable 	| - 			| dmi\_reg\_wr\_en 	| dmi\_wr 	| -  		| -				|-		|

### Benefits of a specified DMI port:

* DTMs can easily reused from other implementations
* The DMI-Bus can be easily swapped out for one that fits better the amount of Debug Moduls

### Suggested Improvement:

DMI-Request signals:
* RequestAddress [`abits`-1:0]
* RequestData [31:0]
* Op[1:0]
* RequestReady
* RequestValid 

DMI-Respond signals:

* RespondData [31:0]
* Respons[1:0]
* ResponsReady
* ResponsValid 

[image]

The survey I have done:
[SurveyRiscVDebugging.pdf](https://github.com/riscv/riscv-debug-spec/files/4574012/LiteratureResearchRiscVDebugging.pdf)

timsifive, 2020-05-06:
DMI is intentionally left unspecified so that people have a lot of flexibility in how to structure their system, and can even reuse their main system bus as DMI if that makes sense. But you raise a good point that in a fairly common case of one DM with one hart directly attached to it, most interface choices between them are arbitrary. It could be helpful for a common implementation in that interface.

I still don't want to add this to the prescriptive part of the specification, since in the end the debugger doesn't care how this happens. But it makes a lot of sense to add a suggested interface to the appendix, to be read in conjunction with the sections that sketch out the possible hardware implementations. If you're interested in writing that up, I'd be happy to work with you to get it in shape to be included.

kmeinhar, 2020-05-07:
I am not familiar with your workflow so please correct me if I have done something wrong.

I have created a work in progress pull request #524 

This one creates a new section in Appendix B.
There a table is given with suggested signals for the DMI.

closed 2020-05-21 by timsifive

jjscheel, 2023-10-27:
Hi, @kmeinhar, I'd like to chat more with you about this commit.  Would you kindly accept my invitation to connect on LinkedIn so we can discuss this contribution more?  Thanks!

PR #524 (2020-05-21): Appendix B suggests signals for a DMI with one DTM connected to one DM
Closes #523