riscv/riscv-debug-spec #525: abstract command feature

gxflying, 2020-06-03:
Hi sir,  I think all the things "abstract command" can do can be done by "program buffer" too,
acctually "abstract command" is always implemented in two ways : 1. several  variable instructions (refer to https://github.com/pulp-platform/riscv-dbg/blob/3670267ab150f5a95e773bc18b1814e4031dc508/src/dm_mem.sv#L331) 2. depending on "program buffer".  Am I right ?
so i want to know why we need a "abstract command" feature, a interface easier to use ? higher execute speed? or something else ?

timsifive, 2020-06-03:
You are right. A few years ago when this was decided, the group was unable to decide whether to use abstract commands or a program buffer. In the end we decided to allow both.

That means you can choose whichever is easiest for you to implement.

gxflying, 2020-06-04:
@timsifive thanks for your kindly reply, this is helpful for me to understand the very original concern of the dbg standard. thanks~

closed 2020-06-04 by gxflying