riscv/riscv-debug-spec #527: the feature about  CSR access is confused,  section 1.4 says required, while chapter 3 says optional ?

gxflying, 2020-06-17:
[image]

pdonahue-ventana, 2020-06-17:
Chapter 1 says that you must to be able to access CSRs.  What you quoted in chapter 3 says that one of the optional ways to access CSRs is abstract access commands.  There's no conflict.

Read down a few lines.  Chapter 3 also says that an implementation must:
> Do at least one of:
> (a) Implement the Program Buffer.
> (b) Implement abstract access to all registers that are visible to software running on the
hart including all the registers that are present on the hart and listed in Table 3.3
> (c) Implement abstract access to at least all GPRs, dcsr, and dpc, and advertise the imple-
mentation as conforming to the “Minimal RISC-V Debug Specification 0.14.0-DRAFT”, instead of the “RISC-V Debug Specification 0.14.0-DRAFT”.

So if you don't do the optional abstract access then you must implement the program buffer (or you can't claim full compliance).

gxflying, 2020-06-17:
thanks!, one more question: 
so as the spec says GPRs must be implemented by abstract access ? and this is obligatory？
but i think in what way to access register is all about the microarchitecture , should we explicitly define here ?

timsifive, 2020-06-23:
> so as the spec says GPRs must be implemented by abstract access ? and this is obligatory？

Yes. You must implement abstract access to the GPRs.

> but i think in what way to access register is all about the microarchitecture , should we explicitly define here ?

We intentionally define only the software-observable behavior of the debug interface, leaving the hardware implementation completely open. There are some hints at possible hardware implementations in the Appendix, although nothing as specific as it sounds like you're looking for.

gxflying, 2020-06-24:
Thanks, it's time to close this issue~

closed 2020-06-24 by gxflying