riscv/riscv-debug-spec #528: behavior of dmcontrol.ndmreset and dmcontrol.dmactive

gxflying, 2020-06-24:
Hi sir:
dmcontrol.ndmreset &  dmcontrol.dmactive, these two bits will  act  the behavior of "reset"
and each "reset" actually  refers to two objects : the core and the Debug Module.
I think there should be some descriptions to **clarify the status of the core and the DM when either execute a "reset"**
but I can not find these informations in the spec which may result in problem on different DM inplementations
a senario 
the **ndmreset**:
this bit will reset all the other parts except for DM ， but it does NOT mention what the DM should do.
when some harts are halted when assert dmcontrol.ndmreset , the core's real state and the state's recoreded by DM will mismatch.
and the debugger will lose control of these cores.  the DM thinks they are still halted while they are NOT. and the current version openOCD expects that the DM should clear the "halt flags" inside DM (see the control flow in deassert_reset)
https://github.com/riscv/riscv-openocd/blob/4f9e2d7171f3cad8d1a99dff3eee5ec5e6d8ea2b/src/target/riscv/riscv-013.c#L2144

timsifive, 2020-11-30:
allhalted/anyhalted in dmstatus always reflect the current state of the hart. Whether it was reset doesn't make a difference. If, at any time, a selected hart is not halted but anyhalted is set that is a bug in the DM. I don't think the spec anywhere says there is a separate halt bit in the DM to track this state.

Section 3.4 explicitly does say:
> Every hart that can be selected is in exactly one of the following four DM states: non-existent, unavailable, running, or halted. Which state the selected harts are in is reflected by allnonexistent, anynonexistent, allunavail, anyunavail, allrunning, anyrunning, allhalted, and anyhalted.

This scenario is no different from when a hart that is halted suddenly (for reasons outside the user's control) starts running.

I'm not sure what to add to the spec to make this more clear.

ernie-sifive, 2020-11-30:
It is up to the debugger to select whether it wants cores to be running or halted following a core reset caused by ndmreset.  The debugger can set haltreq for those cores it wants halted.  Section 3.2 says:

> When a hart comes out of reset and haltreq or resethaltreq are set, the hart will immediately enter Debug Mode
(halted state). Otherwise, if the hart was initially running it will execute normally (running state) and if the hart was initially halted it should now be running but may be halted.

gxflying, 2020-12-01:
thanks @timsifive , it makes sense. as you mentioned , the statsus tracked in dmstatus actually imply the behavior that reset the "halted flags" after a ndmreset , if NOT, the status in dmstatus will NOT match the core's status.  and the the dmstatus.anyhalted bit may always be implemented by "OR" all the halted flags for allthe cores something lile : !(reg_halted_halt1 | reg_halted_halt2 | reg_halted_halt3 ...))
@ernie-sifive I think you may not get my point,  but thank you all the same for your kindly reply ~

closed 2020-12-01 by gxflying