riscv/riscv-debug-spec #529: Section 5.5.1: tselect - "may" versus "must"

TommyMurphyTM1234, 2020-06-26:
Section 5.5.1 says (*emphasis* is mine)

> Writes of values greater than or equal to the number of supported triggers *may* result in a different
value in this register than what was written. To verify that what they wrote is a valid index,
debuggers can read back the value and check that tselect holds what they wrote.

Should this actually be 

> Writes of values greater than or equal to the number of supported triggers *must* result in a different
value in this register than what was written. 

Otherwise there seems to be no guaranteed way to enumerate the number of triggers available?

The current RISC-V openocd implementation of trigger enumeration depends on this behaviour:

https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv.c#L3382

ernie-sifive, 2020-06-26:
This is a bug in OpenOCD.

A trigger index can be "valid" (i.e. can be read back from tselect) even if there is no trigger implemented at that index.  For each trigger index written to tselect, software must first check that the index can be read back, then it must read tdata1 and verify that tdata1.type is non-zero.  When both are true, then there is a valid trigger at this index.  Software can use the fact that "The set of accessible triggers must start at 0, and be contiguous" to stop the search for triggers as soon as it sees one with tdata1.type=0.

TommyMurphyTM1234, 2020-06-26:
Hi @ernie-sifive - thanks a lot.
That makes sense having read the spec about tdata1.type.
Do SiFive have a fix or do you want me to look at it?

TommyMurphyTM1234, 2020-06-26:
Closed and logged as an openocd issue instead.

closed 2020-06-26 by TommyMurphyTM1234