riscv/riscv-debug-spec #530: [clarify] for dmcs2

gxflying, 2020-06-28:
Hi, sir when reading about the dmcs2 in the spec,  I am kind of confused, I think there may be some more description about this.
below are some tips from my point of view, please help to confirm, thanks:

1.  the field "group“ “hgselect” “grouptype”  of dmcs2 are something like "hawindow/hawindowsel" and
   “hgselect/grouptype”  decide which registers to be write, "group“ decides the value to be writen
2. the exttrigger and harts are related by the group ID， that's,  a exttrigger can only trigger the harts with the same group ID

thanks~

gxflying, 2020-07-02:
on more question,  hope your repy @timsifive 
about the cmd_err field,  debug spec states below,  
is it mean that 
1. the very bit writen 1 will be clear (ie. write 0x2 to clear cmd err : 0x2)
or
2.  writen 1 to clear this field (ie. write 0x1 to clear **any** cmd err )
if it is 1st case,  in what senario need we clear cmd err bit by bit ?   or clearing cmd err this way  has other specail concerns?
I think the 2nd case is good enough
[image]

ernie-sifive, 2020-07-02:
Regarding dmcs2:
1. You are right that the protocol to write and read group assignments is indirect, sort of like hawindow.  In the hardware, there will be some kind of "participate" register for each hart and trigger that stores which groupID it belongs to.  You can read and write elements of "participate" through dmcs2.

2. When any hart in a halt group halts, other harts will be halted and all external triggers assigned to that group are notified. Other groups are not affected.

3. Yes, that is correct. The group field when dmcs2 is read will indicate the groupID that the hart selected by hartsel is assigned to.

The purpose of dmcs2, halt/resume groups, and external triggers is to synchronize hart execution with external hardware. You could hook triggers from multiple debug modules, non-RISCV processors, or other logic into a programmable central trigger matrix to have full flexibility over what events in your system will cause harts to halt or resume.  Or it could be as simple as connecting a wire from a JTAG connector to an external trigger input to allow a probe to halt the hart just by asserting a "break" signal.  The ARM connector calls this signal "DBGRQ".

ernie-sifive, 2020-07-02:
Regarding cmd_err:

There are two write values that the spec requires your hardware to support:
1. 000: This has no effect on cmd_err when written.
2. 111: This will set cmd_err to 0 when written.

What happens when you write any other value to cmd_err is not covered by the spec.  Writing 010 might do nothing, it might clear just bit 1, or it might clear all the bits.  Any of those behaviors would be compliant.

gxflying, 2020-07-02:
@ernie-sifive thank you very much for your kindly answers

closed 2020-07-02 by gxflying

gxflying, 2021-01-21:
I have some more questions below,  hope your reply~
1. the dmexttrigger in dmcs2, 
spec says " **If a non-existent trigger value is written here, the hardware will change it to a valid one or 0 if no DM external triggers exist**."
so the exttrigger index should starts from 0 or 1 ?
if starts from 0 and we get a "0" value of the dmcs2.dmexttrigger,  we do not know wheather the system has a exttrigger or doesn't have any at all
2. the setresethaltreq/clrresethaltreq in dmcontrol
 the spec says "**This optional field writes the halt-on-reset rquest bit for all currently selected harts**"
 how to understand the "currently selected harts" ?   and what is the behavior for the flow below:
assume  two harts in the system
1.  select hart 0 only
2.  set setresethaltreq=1
3.  select hart 1 only
4.  reset the two harts
then should the hart0 or the hart1 be halted ?

ernie-sifive, 2021-01-21:
1. You can tell whether there are any external triggers by attempting to write dmcs2 with a 1 in the hgselect field.  This will read back as 0 if there are no external triggers.
2. Assuming in step d that the reset is done externally somehow, then the fact that hart 1 is selected at the time of the reset is irrelevant.  Only hart 0 will be halted after reset.  If instead, in step d you write a 1 to dmcontrol.hartreset, then only hart 1 will be reset and the action will result in no harts halted.