riscv/riscv-debug-spec #541: Software Assert Implementation - Help Needed

micnaelyuanfeng, 2020-08-24:
Hi All,
I am implementing assert function which can cause break in GDB for software debug purpose, like assert in software, cause a trap, GDB will break in Windows Debugger. I tried ebreak in my software, GDB does break in, but after clicking continue, it looks like hart stuck at this line and caonnot move forward. What is correct way to implement an assertion (lib like) to cause GDB break in and continue.

Thanks in advance

ernie-sifive, 2020-08-24:
To move forward, gdb will have to remove the breakpoint, step one instruction, restore the breakpoint, then continue.

micnaelyuanfeng, 2020-08-24:
:) Thanks for reply,

What I did is asm ("ebreak"), so in gdb breaks in. I tried jump *($pc + 4), but it will lock to to asm("ebreak"), do I mess something?

micnaelyuanfeng, 2020-08-24:
Is there any interrupt bit I need clear?

ernie-sifive, 2020-08-24:
gdb should handle everything needed when exiting the breakpoint trap handler.  I am not a gdb expert and I don't know why jump *($pc + 4) didn't work unless the code execution returned to your assert function later.

Are you sure the ebreak opcode asm produced is 4 bytes?  There is also a 2-byte version of ebreak, in which case jump *($pc + 2) would be needed.  You could also try set $pc = $pc + 4 (or 2) and then continue.

micnaelyuanfeng, 2020-08-24:
Thanks !
What I tried as this  asm("ebreak") and gdb breaks in.
I tried -exec jump *($pc + 2) or -exec jump *($pc + 4) in GDB
and then it look back in  ams("ebreak")

> info registers
......
pc             0xd554	0xd554

-exec jump *($pc + 4)
Continuing at 0xd558.

But it goes to this line again
Program received signal SIGTRAP, Trace/breakpoint trap.
251	  asm ("ebreak");

timsifive, 2020-08-24:
The usual way I do this is I create a separate function that just returns instantly. Let's call it break_to_gdb(). Then when loading the program with gdb, I set a software breakpoint in break_to_gdb(). Then gdb knows about the breakpoint and will handle running off it as with any other breakpoint. Then whereever you want your program to break to gdb, just call break_to_gdb().

micnaelyuanfeng, 2020-08-29:
Hi Tim
Thank you very much for your suggestion. I have tried you solution  and it works. Thank you again.
If I want to use ebrake to do assert, do I miss anything? like control register I need to program before I call ebrake in software

timsifive, 2020-09-09:
I would define assert as:
```
void assert(int success) {
   if (!success)
      break_to_gdb();
}
```

micnaelyuanfeng, 2020-09-10:
Thanks Tim,
In the function of break_to_gdb(), I just put  asm ("ebreak"); ? I tried before as I listed the last comments. It keeps send SIGTRAP even I step over, do I miss something except ebreak?

Thanks in advance

timsifive, 2020-09-10:
Don't put any ebreak instructions in your program. Let gdb set them as regular breakpoints, and it won't get confused.

micnaelyuanfeng, 2020-09-11:
Sure. Thanks :) Very appreciate your help

pdonahue-ventana, 2020-10-27:
Should this be closed?

closed 2020-11-03 by timsifive