riscv/riscv-debug-spec #544: timing=before required to perform load?

scottj97, 2020-09-03:
The 0.14-DRAFT spec has this statement under the description of mcontrol.timing=0:

> If this is combined with load then a memory access will be performed (including any side effects of performing such an access) even though the load will not update its destination register. Debuggers should consider this when setting such breakpoints on, for example, memory-mapped I/O addresses.

This change was part of #478 but I don't see any discussion of it there.

The language "**will** be performed" sounds like a requirement that an implementation must follow, but this seems rather onerous as typically, a load with an exception should not perform non-idempotent loads (i.e., to device space).

Could we change that statement to "may be performed" to allow implementation flexibility?

pdonahue-ventana, 2020-09-03:
Shouldn't that warning only apply to data matches?  You can't determine if you have a data match without performing the load.  Of course, address matches on loads can (and I argue should) be done before performing the memory access.

scottj97, 2020-09-03:
Good point, I was considering only address matches. So how about:

"If this is combined with load **and select=1** then..."

pdonahue-ventana, 2020-09-16:
Fixed in the latest spec.

closed 2020-09-16 by ernie-sifive