riscv/riscv-debug-spec #545: Match address trigger with timing=0 and WFI

silabs-oysteink, 2020-09-08:
If a trigger module with timing=0 matches on an address containing a WFI instruction, what happens?

Should the WFI execute as normally when exiting debug mode,  or should it be treated as a NOP as for single step?

ernie-sifive, 2020-09-08:
When the trigger fires, the WFI will not have executed yet.  If you then return from debug mode, the trigger will fire again.  If you clear the trigger and then return from debug mode, the WFI should execute normally.  But, your system will be compliant either way -- WFI is always allowed to act as a NOP.

OpenOCD will do a single-step (with trigger disabled) followed by a resume if the debug cause indicated a trigger breakpoint.  This is done so that you can continue after a trigger without manually clearing the trigger yourself.

silabs-oysteink, 2020-09-08:
I guess the same applies if it matches on an address containing an illegal instruction then? PC will not be set to the exception handler prior to entering debug mode. Clearing trigger and exiting from debug mode will cause the exception to fire.

ernie-sifive, 2020-09-08:
Yes, that is what I would expect.

closed 2020-09-30 by silabs-oysteink