riscv/riscv-debug-spec #550: Interaction of mcontrol.hit with mcontrol.chain?

scottj97, 2020-09-14:
If a trigger with `mcontrol.chain=1` matches, but the next trigger does not match, no exception happens.

Should the first trigger's `mcontrol.hit` bit be set in that case?

ernie-sifive, 2020-09-14:
Yes, I believe so.  The spec says "If this bit is implemented, the hardware sets it when this trigger matches."  When a chain is configured, the triggers compute their own match result independently, then if all match, the action of the last trigger is performed.

scottj97, 2020-09-14:
I suppose that's clear enough. No spec change needed.

scottj97, 2020-09-18:
There's a similar issue with the opposite case: the first trigger in the chain does not match, but the second one would. The description of chain=1 says "While this trigger does not match, it prevents the trigger with the next index from matching."

Combined with the description of hit ("If this bit is implemented, the hardware sets it when this trigger matches") suggests that the second trigger should not set its hit bit, even if it would have absent the chain.

But that's a little different than what @ernie-sifive said above.

ernie-sifive, 2020-09-18:
Your interpretation seems valid.  The text in the Spec suggests that the `hit` bit of the second trigger in a chain should not indicate a match if the first trigger did not match.

scottj97, 2020-11-02:
Since there's a lot of good discussion around the debug spec lately, I'd like to revisit this issue.

The behavior described above ("temperature coding") is rather vague in the spec, and I wonder if it is actually deliberate (and desirable?) or just an accident.

Would it be acceptable, or could we modify the spec to make it acceptable, for the chained triggers to set all their hit bits at once, only when the entire chain matches?

timsifive, 2020-11-03:
This could lead to trouble. If hit is set even when the entire chain doesn't match, you could end up with all the hit bits being set even if the entire chain never matched. Now the debugger cannot use the hit bit to see which trigger hit.

scottj97, 2020-11-03:
@timsifive that's not how I interpreted it. Each trigger in the chain that does not match prevents the next from matching (recursively). The result is a temperature coding where the first 0-N hit bits will end up set. All N will be set if the entire chain matched and a trigger exception was taken. Otherwise, only the first one(s) will possibly be set, which could allow the debugger to see how "close" it came to triggering the breakpoint exception.

This is possibly useful, but seems accidental, which is why I want to have the discussion.

From my selfish hardware design perspective, I would prefer that all hit bits are set only them the entire chain matches; otherwise none of the hit bits in the chain get set.

timsifive, 2020-11-05:
Ah, reading more closely, hit gets set "when this trigger matches" and chain=1 "prevents the trigger with the next index from matching" so I agree with your reading.

I had assumed this would be simpler to implement than setting hit for the entire chain. It could be interesting for a debugger to know how close a chain came to matching, but that seems like a pretty niche case that I cannot think of a real use for. Is it always going to be simpler to implement all-or-none hit being set? How significant of a difference is this?

Right now I'm inclined to leave it as is.

scottj97, 2020-11-05:
This would be the only case where the hit bit gets set absent a breakpoint exception. Without the temperature coding, that never happens.

I'm not sure it would always be simpler. Is there a simple wording change that could make the temperature coding optional?

Data point: WD's SweRV-EH2 supports chains and hit bit, but does not do the temperature coding.

timsifive, 2020-11-11:
#593 merged

closed 2020-11-11 by timsifive

markuslauterbach, 2022-11-18:
@timsifive 
I think there is still a bit of ambiguity of  "match" vs "fire".

I think "match" is quite clear: a trigger matches if its condition matches and if all preceding triggers that are part of the same chain (if there are any) match as well.

But I am not quite sure about the meaning of "fire". It is clear what "chain fires" means but not what "trigger fires" means. A chain fires if all its triggers match. So here "chain fires" is equivalent to "the action of the chain is taken". But when does a trigger fire?

Say, we have a 3-trigger-chain with elements 0-1-2, all triggers match and the chain fires:
trig[0]: match=1, fire=?, hit=?
trig[1]: match=1, fire=?, hit=?
trig[2]: match=1, fire=1, hit=1

A): One possible interpretation is that "trigger fires" means "the trigger's chain fires", which means fire[0]=fire[1]=hit[0]=hit[1]=1.

B): Or an alternative interpretation could be that "trigger fires" means "trigger's action is executed". As only the action of the last trigger in a chain should be taken, this would mean fire[0]=fire[1]=0, so hit[0]=hit[1]=undefined.

Reading through the above thread I assume you intend option A), as this would allow to have "all-or-none hit". However I just saw a target that implements B) with hit[0]=hit[1]=0, and I could not find anything in the spec that forbids this. The only thing that I found is that execute/load/store are defined as "the trigger fires" instead of "the trigger matches", which would indicate that each trigger fires if the chain fires, but this detail can easily be missed or re-interpreted.

As Tim mentioned in most cases a debugger only wants to know if a chain fired, and for this it can simply only look at the last chain element's hit-bit. But in case that it ever gets relevant to look at the other hit-bits as well, it would be good to clarify "trigger fires". If we intend option A) then maybe we could add something like "if all triggers in a chain match, then all triggers in that chain fire" to the definition of the chain-bit.

Or maybe I am missing something?

Thanks!

timsifive, 2022-11-21:
You're right that this is ambiguous.

A trigger fires when that trigger's action is taken. So if a chain matches, only the last trigger in the chain actually fires. That is your option B. I think I want to add the following to the Sdtrig section:
> A trigger matches when the conditions that it specifies (e.g. a load from a
> specific address) are met. A trigger fires when a trigger that matches performs
> the action configured for that trigger.

How does that sound?

markuslauterbach, 2022-11-21:
Yes that would make trigger match and trigger fire clear in my opinion.