riscv/riscv-debug-spec #555: Interaction of icount and tcontrol

scottj97, 2020-10-05:
# Spec

When `icount.m==1`: "When set, every instruction completed or exception taken in M-mode decrements count by 1."

For `icount.count`: "When count is decremented to 0, the trigger fires."

When `tcontrol.mte==0`: "Triggers with action=0 do not match/fire while the hart is in M-mode."

# Questions

Use case: debugging M-mode software using icount breakpoints.

When `tcontrol.mte==0`, clearly the breakpoint exception should not be taken regardless of the count.

Should count still be decremented when executing instructions when `tcontrol.mte==0`? It seems the spec requires it. But then what happens to the breakpoint exception? It cannot be taken because of `mte`.

It seems bad to just drop the exception and never take it. The debugger would then run away without ever stopping as expected.

## Options

When we do `mret` which sets `tcontrol.mte==1`, we could take the breakpoint exception at that time. This would imply that the hardware icount register will have to remember the pending trigger.

Alternatively, when executing in M-mode with `tcontrol.mte==0`, stop decrementing count (if `action==0`), since it can't trap anyway.

ernie-sifive, 2020-10-06:
I think your second option is the right answer.

One purpose of `mte` is to prevent triggers to M-mode from occurring within an M-mode trap handler because this would overwrite `mepc` and render the system unrecoverable.

When using icount triggers from an M-mode debugger, clearly the counter can't be enabled until the M-mode debugger performs `mret` to return to the system code.  You don't want icount counting instructions within the debugger.  So I think using `mte` to enable icount both to count and to trigger is the solution. 

This will have the side effect of preventing stepping of M-mode trap handlers from an M-mode debugger, but M-mode trap handlers are already un-debuggable.

agrobman, 2020-10-14:
I think the spec is over protecting here by preventing breakpoints in exception handlers. Yes, breakpoints in handlers prolog and epilogs will destroy CPU state and not recoverable, but  for handlers written in C with prolog/epilog saving/ restoring mstatus/mpc user still can safely set breakponts in the handler bodies. 
Even BP setting at the beginning /end of the handler can be useful sometimes, when user wants to know if a specific exception/interrupt ever happens

scottj97, 2020-10-14:
Wouldn't such interruptible handlers set tcontrol.mte and/or mstatus.mie to enable breakpoints?

This issue is not about the mechanism for blocking breakpoints in exception handlers (section 5.4 in current draft spec) but about what happens to icount when breakpoints are blocked by that mechanism.

agrobman, 2020-10-14:
You are right, this is different issue, but  there are provisions in spec for systems without tcontrol to use mie=0 state to block BPs with action=0, which is weird...  I can't setup HW BP on piece of code which disables interrupts if I use SW base debugger . Or if my application has nothing to do with interrupts...

PowerPC had the same issue more than 30 years ago, which was solved by BookE specification, defining a special pair of "debug" state save SPRs, which were used by HW to save CPU state in case of "breakpoint" exceptions and special RFI flavor instruction to return ..

scottj97, 2020-10-14:
> You are right, this is different issue

Please open a new issue for that discussion.

closed 2020-10-23 by timsifive

scottj97, 2020-10-24:
With the new pending bit added by #574, it seems this behavior (of blocking the count decrement when trigger exceptions are blocked) doesn't seem so important anymore. The exception would no longer get forgotten but instead become pending and get taken ASAP.

commit 99dfc98f (2020-10-23): Don't decrement icount.count when exception is blocked by tcontrol.mte (#557)

* Don't decrement icount.count when exception is blocked by tcontrol.mte

Since it wouldn't be able to take the breakpoint trap anyway.

Fixes #555.

* Expand count decrement exception to include other trap-suppression mechanisms