riscv/riscv-debug-spec #556: Relative priority of etrigger/icount/itrigger/mcontrol(after)

scottj97, 2020-10-06:
Table 5.2 shows synchronous exception priority. Mutually exclusive exceptions of the same priority appear together in one row of the table. For example, no single instruction can have both 15 (store/amo page fault) and 13 (load page fault) so there is no need to define relative priority between them.

The first row of the table includes etrigger, icount, itrigger, and mcontrol(after), suggesting these exceptions are mutually exclusive.

But they are not! It's possible to have icount trigger at the same time as etrigger at the same time as mcontrol(after).

Shouldn't these exceptions have their relative priority defined, then? It matters because which trigger caused the exception will determine what to write into `mtval` (and perhaps more?).

ernie-sifive, 2020-10-06:
Hi Scott,

The draft spec (
https://github.com/riscv/riscv-debug-spec/blob/master/riscv-debug-draft.pdf)
discusses this topic in section 5.3.

The value to be written to mtval is mandated in the Privileged Spec: "When
a hardware breakpoint is triggered, or an instruction-fetch, load, or store
address-misaligned, access, or page-fault exception occurs, mtval is
written with the faulting virtual address."  What to write to mtval is
obvious for breakpoint types that involve an address of some kind. For
other types (icount, etrigger, itrigger), the "faulting virtual address"
would be the address of the next instruction to execute.  For
mcontrol-after set on a data access, it would be the data address.  Is this
the conflict you're asking about?

My opinion is that this is implementation-dependent. But I would think that
the data address would be more useful since the PC is already available in
DPC.  Assuming hit is implemented, the debugger will see multiple triggers
so it will be clear to the user what happened.

Ernie

scottj97, 2020-10-06:
Yes that's the mtval conflict I was referring to.

This table also affects which Hit bits are required to be set. The spec says "When multiple triggers in the same priority fire at once, hit (if implemented) is set for all of them." Which says (among other things) that if an mcontrol-after trigger matched the just-committed instruction, and an icount trigger has just counted down to 0, both of those Hit bits must be set, since they appear in the same box in the table.

I suppose that's desirable behavior and we wouldn't want to change that by defining relative priority among these different breakpoint causes.

I just find it curious that this is the only place where multiple simultaneous exceptions can occur where the spec does not define any priority between them.

closed 2020-10-06 by scottj97