riscv/riscv-debug-spec #559: Details of icount.count field

scottj97, 2020-10-06:
I'm thinking about what happens when an icount breakpoint exception occurs. If the handler does not touch the CSRs, is it acceptable for the trap to reoccur immediately after `mret`? If not, what mechanism prevents this?

The spec describes when `count` should be decremented, and "When count is decremented to 0, the trigger fires."

What if count is already 0? Is it decremented anyway, wrapping around to 0x3fff (or whatever platform-defined maximum)?

If the answer is that it should clamp to 0, what happens when a CSRW writes count=0? Should it take a trap immediately, or is the breakpoint effectively disabled? One could argue that the spec does not allow a trap since writing 0 directly is not "decremented to 0".

ernie-sifive, 2020-10-07:
I think the trigger occurs when count=1 and an instruction or trap occurs in one of the enabled modes.  Once the breakpoint trap is taken, count remains at 0 (or m/s/u stay at 0).  The spec doesn't say, but if your debugger mrets with count=0, I would think count would decrement from 0 to 0x3fff, etc., until it traps again.

If non-debugger code sets count to 0, I would say this should not trap (unless count was 1 just before).

scottj97, 2020-10-07:
When you say "count remains at 0" I assume that's because of `tcontrol.mte=0` (or `mstatus.mie=0`)?

I think there's a practical problem with this approach. Let's say I write count=2, m/s/u = 0/0/1, then mret to U-mode. The first U-mode instruction completes successfully, decrementing count to 1. The second instruction has a load page fault exception that goes to an M-mode handler. The count needs to decrement, since a trap was taken from U-mode, and the count is now 0, but we can't take the breakpoint exception because the page fault has set `tcontrol.mte=0`.

The count will stay at 0 until the mret -- but when/how does the breakpoint exception occur? 0x3fff instructions later? Not good.

If, instead, we took a breakpoint exception any time count was 0, the mret in this case would immediately trap back to the debugger, before running a single U-mode instruction and before decrementing (and wrapping) count, which seems like the most desirable behavior.

That has the drawback that writing count to 0 would also cause a trap when enabled. But maybe that's okay?

scottj97, 2020-10-07:
Is there any implementation of icount triggers that I could refer to? Or a software debugger implementation that can take advantage of icount-type triggers?

ernie-sifive, 2020-10-07:
Offhand, it seems like your approach would work.

timsifive, 2020-10-08:
Decrementing the count should never wrap.

Another solution to your problem is to have the page fault handler execute normally, and when returning to U mode it immediately takes the trigger that was "queued" because it decremented to 0. I'm not sure how much complexity that tracking adds to the implementation. This is kind of what you're doing by using the count value to signal a pending trigger.

Perhaps when the trigger actually fires it should clear m/s/u?

I'm not aware of any icount implementations.

scottj97, 2020-10-19:
Let me throw another wrench into the works.

If an implementation does not have one of the optional mechanisms in 5.4 (tcontrol.mte or mstatus.mie) to block exceptions in M-mode, then my suggestion of triggering when count=0 will not work, since it will result in endless breakpoint exceptions.

Similarly, what if breakpoint exceptions are delegated to S-mode via medeleg? The trigger would still be pending (since count=0) and not disabled (since tcontrol.mte and/or mstatus.mie are still 1) so it would get stuck in endless S-mode breakpoint exceptions.

@timsifive's idea of an internal "trigger pending" bit (which gets cleared by hardware on a CSRW to icount or on an icount breakpoint exception) would solve both of these, but feels unclean.

Clearing m/s/u on the breakpoint exception would not solve these issues, I don't think, because while those bits block decrementing the count, they don't block the pending exception. (Or do/should they?)

It seems, in any case, there are spec changes needed to explain the required behavior.

timsifive, 2020-11-03:
Will be fixed when #574 merges.

timsifive, 2020-11-05:
Fixed by #574.

closed 2020-11-05 by timsifive