riscv/riscv-debug-spec #562: External debugger modification of code

pdonahue-ventana, 2020-10-09:
Nothing really came of https://lists.riscv.org/g/tech-debug/topic/75891547 so I'm filing this issue so it doesn't get lost.

Short summary of the problem: In an implementation that does not implement the program buffer, how does the debugger guarantee that the new ebreak is visible to instruction fetches after writing the ebreak instruction to memory via an abstract command or system bus access?

timsifive, 2020-11-03:
Once the cache management group (https://lists.riscv.org/g/tech-cmo/topics) gets something done, we'll be able to address this properly. It is a glaring omission in the spec right now.

pdonahue-ventana, 2020-11-03:
The CMO group isn't working on coherence between instruction and data caches.  That's either the existing Zifencei extension or the in-progress J extension.  That being said, I think that waiting for 1.1 is fine.