riscv/riscv-debug-spec #565: M-Triggers with action=0 and MIE bit

agrobman, 2020-10-14:
I think the spec is over protecting here by preventing breakpoints in exception handlers. Yes, breakpoints in handlers prolog and epilogs will destroy CPU state and not recoverable, but for handlers written in C with prolog/epilog saving/ restoring mstatus/mpc user still can safely set breakponts in the handler bodies.
Even BP setting at the beginning /end of the handler can be useful sometimes, when user wants to know if a specific exception/interrupt ever happens.

There are provisions in spec for systems without tcontrol to use mie=0 state to block BPs with action=0, which is weird... I can't setup HW BP on piece of code which disables interrupts if I use SW base debugger . Or if my application has nothing to do with interrupts...

PowerPC had the same issue more than 30 years ago, which was solved by BookE specification, defining a special pair of "debug" state save SPRs, which were used by HW to save CPU state in case of "breakpoint" exceptions and special RFI flavor instruction to return ..

timsifive, 2020-10-14:
The spec doesn't actually require anything. It states the problem:

> On full-featured systems this is a remote corner case that can probably be ignored. On systems
> that only implement M mode, however, it is recommended to implement one of two solutions to
> this problem. This way triggers can be useful for native debugging of even M mode code.

And then lists two possible solutions:

> The simple solution is to have the hardware prevent triggers with action=0 from firing while in M
mode and while MIE in mstatus is 0.

> A more complex solution is to implement mte and mpte in tcontrol.

But nothing is required. You can implement neither in a system.

What do you think this section should say?

agrobman, 2020-10-14:
I would offer just a warning that trigger in these conditions can cause processor state lost and users need to be cautious, when debugging interrupt/exception handlers. 
Or propose to have these special CSRs similar to PowerPC BookE for breakpoint exception with special mret ...

I agree that only M-mode systems most likely won't be debugged with SW based debugger - SW based debugger  is for "full-featured systems". So this is a relatively small problem, but now the spec suggests this strange dependence when BP does not fire when interrupts are disabled, which is not that obvious to even  people, closely working with RISCV processors.

scottj97, 2020-10-16:
> I would offer just a warning that trigger in these conditions can cause processor state lost and users need to be cautious, when debugging interrupt/exception handlers.

It already says that (section 5.4 of current draft-0.14 spec):

"When such a trigger causes a breakpoint exception while already in a trap handler, this will leave the system unable to resume normal execution."

agrobman, 2020-10-16:
1) this is too strict statement - system wont loose state if it was previously saved (in stack)
2) I would remove provision to make triggers dependent on MIE ... our designers take it too literally .. :(

timsifive, 2020-12-03:
1. I'll change "will" to "might."
2. I think the spec is clear about the trade-off there. It presents two options, and provides the drawbacks of both. Does somebody want to argue that the more complex solution is still so trivial that we shouldn't mention a simpler solution at all?

closed 2020-12-18 by timsifive