riscv/riscv-debug-spec #568: Clarify tdata1.hit behavior

JamesKenneyImperas, 2020-10-15:
Could you please clarify the behavior of `tdata1.hit` bits, specifically:
1. whether this bit is sticky (I think it is, but this is not stated); and:
2. what precisely is meant by _matches_ in statements of the form _the hardware sets it when this trigger matches_. For example, does a trigger with `action=0` for which the condition is true count as matching if `tcontrol.mte=0`?

Thanks.

ernie-sifive, 2020-10-15:
Hi James,

The *hit* bit is sticky.  The text says that the hardware will set it when
the trigger matches and that the debugger can set or clear it at any time.
When the trigger matches, *hit* stays at 1 until it is written with 0 from
the debugger.

I would say *matches* means all trigger conditions are satisfied.  Normally
this means the action is taken but it might not be if more than one trigger
occurs simultaneously and they have different actions.  As for *mte*, the
text states "*0: Triggers with action=0 do not match/fire while the hart**
is in M-mode.*"  This indicates that, if it is implemented, *mte=1* is one
of the conditions for a match and therefore *hit* is not set if a trigger
matches all conditions other than *mte=1*.

Ernie

tianshi-xty, 2020-10-16:
Hi，ernie

I have another questiong about tdata1.hit bits. 

“If this bit is implemented, the hardware sets it
when this trigger matches. The trigger’s user can
set or clear it at any time. It is used to determine
which trigger(s) matched.”

Assuming that a trigger is not configured at the beginning. Debugger set the tdata1.hit incorrectly when he want to use the trigger.  In the case，need hardware ignore the setting of tdata1.hit？

JamesKenneyImperas, 2020-10-16:
I believe this bit is set by hardware when the trigger fires, but otherwise has no hardware effect - the purpose is simply to inform the debugger. Therefore, setting or clearing this in the debugger will not change the behavior of the trigger.

tianshi-xty, 2020-10-16:
I agree what you said. Setting or clearing this in the debugger will not change the behavior of the trigger. But if debugger set the tdata1.hit bit of trigger incorrectly, the tdata1.hit bit will give debugger a wrong information when other trigger fire.

For example, there have two trigger, trigger_0 and trigger_1. Debugger may consider that he has written 0 into tdata1.hit of trigger_0. But actually he written 1 into tdata1.hit bit of trigger_0. When trigger_1 fires, but trigger_0 not fires,  Debugger will get wrong information that trigger_0 and trigger_1 both fires. 

So if tdata1.hit is 0, need hardware ignore setting tdata1.hit bits to 1？

ernie-sifive, 2020-10-16:
As James said, setting _hit_ has no effect on the hardware.  This is one of many examples of where a debugger could cause problems for itself or for the hardware if it writes the wrong thing to the wrong place.

tianshi-xty, 2020-10-16:
Now I unserstand . Thank you!

closed 2020-11-03 by timsifive