riscv/riscv-debug-spec #569: Clarify data value for triggers on sign-extending loads

JamesKenneyImperas, 2020-10-15:
For a data value trigger on a sign extending load, what value is used for value comparisons? For example, if an `lb` instruction loads the value `0x80` from memory, is the data value considered to be `0x80` or `0xffffff80` for the purposes of the trigger?

Thanks.

ernie-sifive, 2020-10-15:
Hi James,

I think the number of bits of *tdata2* that are compared to the load value
is determined by the *size* field.  If you have a trigger conditioned by
*size=1*, then only 8 bits are compared.  As stated in the text, *size=0*
with *select=1* (data value trigger) has undefined behavior.

Ernie

JamesKenneyImperas, 2020-10-15:
Hello Ernie,

Are you saying that, in the example I gave, only the low 8 bits of the loaded value would be compared with the lower 8 bits of tdata2?

If so, can you please also clarify what the _lower half_ and _upper half_ references mean in match types 4 and 5? Are these 4-bit nibbles in this case, or something else?

Thanks.

ernie-sifive, 2020-10-15:
Yes, only 8 bits should be compared if size=1.

The spec is a little vague on the masking definition.  I would take it to mean that the 8-bit data value is extended to XLEN (doesn't matter if zero-extended or sign-extended), then the mask is applied, then part of the result is compared to part of the value according to `size`.  The mask is the upper half of tdata2 (16 bits with XLEN=32 or 32 bits with XLEN=64) and the value is the lower half of tdata2.

So in your example with size=1 and XLEN=32:

- match=4 means `trigger = (value[7:0] & tdata2[23:16]) == tdata2[7:0]`.
- match=5 means `trigger = 1`.  (Upper 16 bits are don't-care when size=1)

JamesKenneyImperas, 2020-10-16:
Thanks for the clarification. I think it would be helpful to state this behavior explicitly in the specification - I'll leave this issue open as a reminder.

tianshi-xty, 2020-10-28:
Hi Ernie，

“match=5 means trigger = 1. (Upper 16 bits are don't-care when size=1)”
Can you please explain why trigger = 1 when upper 16 bits are don't-care when size=1？If we don't care about  upper 16 bits of datavalue, is it more reasonable to make trigger = 0?

JamesKenneyImperas, 2020-10-28:
I think Ernie's second example probably isn't right; assuming `datavalue[7:0]` is zero-extended to 32 bits, the logic is:

`trigger = ((datavalue[23:16] & tdata2[23:16]) == tdata2[7:0])`

and given `datavalue[23:16]=0`, this is simply:

`trigger = (0 == tdata2[7:0])`

closed 2020-12-21 by timsifive