riscv/riscv-debug-spec #570: Clarify trigger priority for AMO instructions

JamesKenneyImperas, 2020-10-15:
Table 5.2 gives the priority of triggers with respect to standard exceptions. Could you please clarify how this works with AMO instructions other than `amoswap`?

For such instructions, the value stored by the instruction is not known until after the load. This means that a _mcontrol/mcontrol6 store data before trigger_ cannot fire at the indicated point but must happen later. I think the table needs to be enhanced to discuss this case, and also to explicitly state whether `amoswap` behaves differently to other AMO instructions in this respect.

Thanks.

ernie-sifive, 2020-10-15:
I think triggers on AMO data value, whether load or store, typically have timing=1 (after) if a system can support AMO store data value triggers at all.  The AMO operation may be performed in the memory system so the core may not ever see the store value.

Assuming the trigger unit sees both the load and store value, then with timing=1 the priority is clear from Table 5.2 -- the mcontrol exception is taken as the highest priority.

Supporting timing=0 with select=1 is complicated so not typically supported.  Store data value is uncomplicated as long as the system doesn't support the A extension.  Otherwise, if the breakpoint/debug exception ends up being taken, then the hardware must revert the system to the state it was in before the triggering instruction, including restoring any changed memory contents and registers.  If the spec needs to address such systems, we could update Table 5.2 to replace "mcontrol store data before" with "mcontrol non-AMO store data before" and add "mcontrol AMO store data" at the same priority as "mcontrol load data before".

Ernie

JamesKenneyImperas, 2020-10-16:
Thanks Ernie. I think text based on your response here should be included in the specification near table 5.2, especially the point that _triggers may not be supported for AMO operations at all_ - as far as I can see, AMO operation constraints are not discussed anywhere in the specification, and this came as a complete surprise to me. From my naive viewpoint, setting a breakpoint on a lock operation sounds like something that users might regularly attempt to do, so it should be made very clear that this might not be possible.

pdonahue-ventana, 2020-10-16:
What are your thoughts on https://github.com/riscv/riscv-debug-spec/pull/561?

JamesKenneyImperas, 2020-10-16:
I'm not an expert on debug hardware at all, but that change looks to be starting to address this issue, when combined with Ernie's comment above.

Given that all of this looks to be very implementation specific, I am wondering whether in such situations it would help to take a leaf out of Arm's book in the way that it is specified. They have the concept of _constrained unpredictable_, where the exact behavior is implementation specific, but must be one of a fixed set of choices which are explicitly listed.

timsifive, 2020-11-13:
I think any 1.0 behavior changes we might make are done. @JamesKenneyImperas, are more clarifications required, or can this be closed?

JamesKenneyImperas, 2020-11-14:
Yes, I think that looks sufficient. Closing.

closed 2020-11-14 by JamesKenneyImperas