riscv/riscv-debug-spec #571: Specification version confusion

JamesKenneyImperas, 2020-10-20:
I'm confused as to what the version specification 0.14.0-DRAFT is intended to mean.

I have downloaded two pdf files from the Latest Draft link here, one in July and one a few days ago. Both claim to be 0.14.0-DRAFT, but they are different: the more recent one describes extra support for Virtualization and modifies the `scontext` CSR address, for example.

What is the policy for update of the version number when the Latest Draft link is updated? I think it would be better if every such update produced a new version to avoid any ambiguity.

ernie-sifive, 2020-10-20:
Hi James,

I think the idea is these are all drafts for the upcoming version 0.14.0
that will be submitted for ratification.  You can differentiate them by the
date or git hash on the title page.  Tim Newsome rebuilds the PDF every few
weeks to get the latest changes.  You can star the github repository (
https://github.com/riscv/riscv-debug-spec) to get notified of updates.

The current plan is to freeze the content about Nov. 15 and complete the
ratification process in March.

Ernie

JamesKenneyImperas, 2020-10-20:
Hi Ernie,

Thanks for the clarification. I'll close this issue and ask customers who refer to "version 0.14.0-DRAFT" to be more precise.

James.

closed 2020-10-20 by JamesKenneyImperas