riscv/riscv-debug-spec #572: Clarify chaining of triggers of type 3, 4 and 5

JamesKenneyImperas, 2020-10-22:
The specification gives examples of trigger chains using type 2 (or 6) triggers. Types 3, 4 and 5 have no `chain` field. Can you please clarify the specification for triggers of these types in this regard? Specifically:

1. Is it legal to terminate a chain with a type 3, 4 or 5 trigger?
2. If it is legal, does the `timing` of all previous entries in the chain have to be `after` for the chain to match?
3. Can you enhance the documentation to include an example where this might be useful?

Thanks.

timsifive, 2020-11-03:
The intent was to only chain type 2 (or 6) triggers, and not to mix them with different types. I'll clarify that later.

timsifive, 2020-11-13:
@JamesKenneyImperas, can this be closed?

JamesKenneyImperas, 2020-11-14:
Yes, closing.

closed 2020-11-14 by JamesKenneyImperas