riscv/riscv-debug-spec #573: Is tdata2 banked by trigger type?

JamesKenneyImperas, 2020-10-23:
The descriptions for type 4 and 5 triggers each have a sentence of this form:

> Hardware may only support a subset of interrupts for this trigger. A debugger must read back tdata2 after writing it to confirm the requested functionality is actually supported.

Does this mean that `tdata2` is banked per trigger type? For example, consider the following sequence:
1. write `tdata1` to select type 2 trigger (`mcontrol`);
2. write all-ones to `tdata2`;
3. write `tdata1` to select type 4 trigger (`itrigger`);
4. read `tdata2`.

What value is seen by the read in step 4? Is it:
1. all-ones, as was written?
2. a subset of all-ones, masked by the subset of supported interrupts? (in which case, what is read if the trigger type is set back to 2 again?)
3. an unknown value, because `tdata2` is banked per trigger type?
4. something else?

Thanks.

ernie-sifive, 2020-10-23:
The spec says this:

1. All tdata registers follow WARL semantics.
2. If a debugger writes an unsupported configuration, the register will read back a value that is supported.
3. Writes to one tdata register may not modify the contents of other tdata registers.

In your example, step 3 may result in tdata1 either not being changed or being changed to a disabled trigger since all-ones is not a legal value for tdata2 in itrigger mode.  Per Rule 3, tdata2 will not be changed by step 3 and will read the same value after step 4 as it would after step 2.

pdonahue-ventana, 2020-10-24:
The term "may not" in rule 3 is ambiguous.  Every other usage of the term "may not" in this document appears to be synonymous with "might not" (or simply "may") but I "may not" be interpreting it correctly. 😄

pdonahue-ventana, 2020-10-24:
How about this sequence:
1. write tdata1 to select type 2 trigger (mcontrol);
1. write all-ones to tdata2;
1. write tdata1 to select type 4 trigger (itrigger);
1. write 0 to tdata2
1. read tdata1

Is it legal for the read to return type 4?  At the time of the read, that is a legal value and WARL simply requires legal reads.

JamesKenneyImperas, 2020-10-24:
I think the interlocking behavior of `tdata1` and `tdata2` as described here is not obvious and makes programming and deprogramming triggers a tricky task. Given the description above, it seems that code that wants to change the programming of a type 2 trigger to a type 4 one in one step must change `tdata2` first, followed by changing `tdata1`, otherwise the attempt to change `tdata1` could fail. However, changing a type 4 trigger to a type 2 one in one step requires changing `tdata1` first, otherwise the write to `tdata2` will in all probability fail. Is this really the case, or have I misunderstood something?

If it really is like this, then I think the documentation should be enhanced to give recommended sequences for updating triggers  safely and correctly. From Ernie's initial response, it seems that a type 0 (disabled) trigger allows any value in `tdata2`, in which case this sequence would I suppose always work:
1. write `tdata1` to select type 0 trigger (disabled);
2. write `tdata2` to the required value for the about-to-be-selected trigger type;
3. write `tdata1` with the type 2, 3, 4, 5 or 6 trigger type.

It would also be worth mentioning in such a section that trigger updates in Machine mode (using triggers for non-Debug purposes) should be sure to either set `tcontrol.mte=0` or `mstatus.MIE=0` (depending on what is implemented) if those triggers are activated in M-mode, otherwise great care is needed to avoid partially-programmed triggers firing during the programming process, especially if they are chained.

timsifive, 2020-11-12:
A debugger should be able to write a trigger by writing first tdata1 (and reading back), then tdata2 (and reading back), etc.

As shown here, that's not the case with the current spec. I believe the following does always work:

1. write 0 to tdata3
2. write 0 to tdata2
3. write desired value to tdata1
4. read back tdata1 to check it's supported
5. write desired value to tdata2
6. read back tdata2 to check it's supported
7. write desired value to tdata3
8. read back tdata3 to check it's supported

The more I think of it, the more I dislike the WARL nature of the trigger register. Everything is so much simpler if they're just DXLEN-bit registers that get interpreted by the hardware. If we had a configuration structure for this data we could use that, but we don't. It seems too late to change that right now.

Another option (mentioned in #583) is to have writes to tdata1 clear tdata2 (and presumably tdata3). That makes life easier for the debugger, but harder for the hardware. Generally we try to push complexity into the debugger.

I'm inclined to update the spec to tell debuggers the general way to write triggers as I mentioned. (I feel less certain about @JamesKenneyImperas's sequence.) Thoughts?

Sorry to get to this so late. FIFO is fair, but sometimes you end up not seeing the important issues until it's too late.

ernie-sifive, 2020-11-12:
I think the WARL semantics does work, but I think it would be useful to enumerate the recommended programming sequence.  I really don't think you ever want a write to one CSR to change the value of another CSR.

Trigger programming software must already follow this sequence, more or less.  I do know of one incident where software first wrote to tdata1, then tdata2 and the engineer was surprised that a trigger occurred between the two.  Whenever there are multiple registers involved, software must be careful to program them in a sequence such that there is no possibility of spurious actions between writes.

To your sequence, I might add a step 0 which would be to disable the trigger by writing 0 to M, S, and U bits.  Otherwise, a spurious trigger could occur between step 2 and 3 if the running software accessed address 0.  Also, keep M, S, and U at 0 until after step 8 or you risk spurious triggering on an access of address or value of 0 between step 3 and 5.

JamesKenneyImperas, 2020-11-12:
I'm also not a fan of the WARL semantics. There are plenty of precedents elsewhere for a write to one CSR affecting the value of another (for example, mideleg affects the value of sie/sip) so this wouldn't be unusual.

If it is considered to be too late to change this, then I would recommend that descriptions of tdata1, tdata2 and tdata3 should be augmented to explicitly state all cases in which writes to those registers may fail because of conflicting state in other registers. This will help people implementing the CSR write logic.

timsifive, 2020-11-12:
Can we specify that writing 0 to tdata1 must always end up with that trigger completely disabled? I think that's already true on existing implementations (although I don't know about ones where tdata1.type is writable), and then debuggers can set a trigger as follows:

1. Write 0 to tdata1.
2. Write relevant values to tdata2 and tdata3.
3. Write configuration to tdata1.

That seems pretty reasonable. Thoughts?

JamesKenneyImperas, 2020-11-13:
Hello Tim,

I think this is exactly the sequence I proposed in the fifth post above. It works if it is explicitly stated that tdata2 and tdata3 can hold _any legal value for any trigger type_ if tdata1 has value 0.

I still think it is important to also explicitly state all cases in which writes to tdata1, tdata2 or tdata2 may fail because of conflicting state in other registers.

James.

pdonahue-ventana, 2020-11-13:
I think that there's a difference between what Tim said ("Write 0 to tdata1") and what James said ("tdata1 has value 0").  Implementations are allowed to only support, for example, type=4 when tselect=3.  So the type field (and others) may be hard-wired to non-zero values.  Or it may support both type=4 and type=5 so only part of the type field is hard-wired.

I didn't think through the implications of this in this context but thought I'd point it out.

JamesKenneyImperas, 2020-11-14:
Hi Paul,

Yes, good point. Well spotted!

James.

closed 2020-12-03 by timsifive