riscv/riscv-debug-spec #582: textra32/textra64, sselect=2, VU/VS mode

JamesKenneyImperas, 2020-11-02:
Section 5.5.16 says, when `sselect`=2:

> This trigger will only match if ASID in `satp` equals the lower ASIDMAX (defined in the Privileged Spec) bits of `svalue`

This does not seem useful in VS/VU mode. In that case, the ASID is determined by `vsatp`, not `satp`.

gfavor, 2020-11-02:
While it is true that the value in vsatp is the active value when in VS/VU modes, it is also true that vsatp replaces satp, for all intents and purposes, when in VS/VU modes (e.g. CSR accesses to satp actually read and write vsatp).  From an architectural perspective vsatp appears as or in place of satp in VS/VU modes and any uses in the architecture of satp actually use vsatp.  The Debug spec should be no different in this regard - it should always be using the active "satp" value (oirrespective of whether it comes form hte physical satp or vsatp register).  The Privileged spec takes this same approach, i.e. it is understood that references to 'satp' in fact become references to 'vsatp' when in VS/VU modes.

JamesKenneyImperas, 2020-11-02:
It is of course true that in VS/VU modes `vsatp` behaves like `satp` does in HS/U modes, but it is not the same register. I don't think it would do any harm to clarify the required behavior here and state exactly which ASID value is compared under which conditions.

It would also help to explicitly describe the behavior for HLV, HLVX and HSV instructions, which presumably use `vsatp.ASID` for the comparison.

gfavor, 2020-11-02:
Clarification is often a good thing (even though some would argue that this behavior is "clearly" implied by the Privileged spec - which loves to be on the concise and terse side).  I'm not arguing for or against clarification in this case.

As far as the H* instructions, they are defined to "perform an explicit memory access as though V=1" - which means that everything MMU, PMA, and PMP wise (and all other stray bits of of existing and new architecture) is handled as if V=1.  As far as MMU-related functionality, all the vs* state becomes active for the sake of performing an H* instruction.  This is essentially the general principle that applies to all *atp and ASID usage when V=1.

But again, if clarification in the debug spec of this particular point is felt to be warranted, then something like this could be said:  "The currently active ASID value, from either satp or vsatp, is used."

JamesKenneyImperas, 2020-11-02:
Actually it has occurred to me that HLV, HLVX and HSV instructions are more subtle, because although they _access memory_ as if V=1 they are _executed_ with V=0. Therefore, for these instructions, `satp.ASID` should be used for _execute address_ breakpoints and `vsatp.ASID` for _load/store address breakpoints_.

Will anyone ever want to specify a breakpoint on an HLV instruction that is executed in HS mode with `satp.ASID=0x1` that accesses VS memory for which `vsatp.ASID=0x2`? That isn't possible at the moment, because there is only a single `scontext` register to compare against.

timsifive, 2020-12-17:
> Will anyone ever want to specify a breakpoint on an HLV instruction that is executed in HS mode with satp.ASID=0x1 that accesses VS memory for which vsatp.ASID=0x2? That isn't possible at the moment, because there is only a single scontext register to compare against.

Regardless of scontext, a certain type of instruction performing some memory access would require chaining two triggers. Each trigger gets its own textra{32,64} so they can have separate sselect values.

closed 2020-12-19 by timsifive