riscv/riscv-debug-spec #583: Is it too late to modify tdata register formats and semantics to improve usability?

JamesKenneyImperas, 2020-11-02:
I want to start this issue with an apology: I have only recently looked at the Trigger Module part of this specification in any detail, so the points I'm making here should really have been raised a long time ago - I'm sorry about that. 

I am very concerned that the current specification of the trigger registers, especially `tdata1` and `tdata2`, makes their use very unwieldy. Clearly these have evolved significantly over time (I have not followed any of the history of that) but I think the resulting specification will cause unnecessary difficulties for all future implementations and programmers. If at all possible, I think this should be revised before submission for ratification.

I think that when designing a CSR interface, several principles should be observed:

1. Wherever possible, field positions and meanings should match for all XLEN. In this specific case, it would be better to locate the `type` and `dmode` fields in the lower 32 bits of the `tdata1` register, not at the top. This makes it much easier to write code that can be compiled for any XLEN and still work.  The one exception to this is that it sometimes makes sense to put a flag at bit XLEN-1 to facilitate branching.

2. When a register has different views (like `tdata1`), fields with the same meaning should appear at the same location so that it is easy to write common code for all views. This isn't true for many of the fields in `tdata1`: the worst offender is the `hit` bit, which can be found at positions 20, 22, 24 or XLEN-6, but many fields have at least two different positions, depending on view.

3. As a minor point, where fields have analogues in existing CSRs, it makes sense to locate them in the same place. For example, the `m`, `s` and `u` mode bits might be more logically placed at bits 3, 1 and 0 where they line up with similar mode bits in `mie`, but aligning the whole group to any nibble would help.

4. The interface should be easily programmable without pitfalls for the unwary. The semantic definition mentioned in issue #573 seems to interlock `tdata1` and `tdata2` in an undesirable manner, with each only being writable if the other is in a particular state. I think the rules listed there should be modified: perhaps any write to `tdata1` that changes `type` should automatically clear `tdata2`?

5. The implications of operations with different XLEN need to be considered and clearly stated. It isn't clear to me how exactly the triggers behave when `DXLEN=64` and `MXLEN=32`, for example, especially when both M mode and the debugger are independently programming triggers for their own purposes.

Once again, I apologise for raising these things so late in the day. Please feel free to close this issue if it is too late to do anything, or if all these things have previously been addressed in other issues.

pdonahue-ventana, 2020-11-02:
Now that you mention it, this also becomes a problem for CSR width modulation as described in section 2.4 of the privileged architecture.  If XLEN changes from 64 to 32 then weird things happen to the type and dmode fields.

timsifive, 2020-11-03:
1, 2, 3: I agree, but for the sake of backwards compatibility I'd like to just keep things the way they are. There are software/hardware implementations of a good chunk of triggers already.

4: Something should be done about this.

5: Needs a little bit of thought also.

timsifive, 2020-11-12:
Let's discuss 4 in #579 

Here let's talk about the "implications of operations with different XLEN." The spec already says:
> In this section XLEN means MXLEN when in M-mode, and DXLEN when in Debug Mode. Note
> that this makes several of the fields in tdata1 move around based on the current execution mode
> and value of MXLEN.

That's clear to me. Is it clear to you?

I don't see any other obvious problems, either, but I bet I have blind spots. @JamesKenneyImperas, can you elaborate on what is not clear to you?

JamesKenneyImperas, 2020-11-12:
Some examples to consider below.

1. Suppose that MXLEN=64 and textra64.svalue is programmed with a value with more than 16 significant bits, or textra64.mhvalue is programmed with a value with more than 6 significant bits for a trigger causing an M-mode breakpoint exception. What happens if MXLEN changes to 32? Are the most significant bits of svalue/mhvalue discarded, or ignored, or something else? What value is seen in textra64 if MXLEN is set back to 32 again?

2. I think that the meaning of _move around_ in the statement above needs clarification. For example, suppose that DXLEN=64, MXLEN=32, and that the debugger wants to program a trigger to activate itself on an M-mode event. I presume it always updates the trigger registers using DXLEN=64 format; if so, what values are seen if those registers are _read_ by M-mode code? Is it the lower 32-bits of the DXLEN=64 format value (which is likely to be illegal because field positions are not the same in the two views) or a dynamically-reordered register view (in which case, what is seen for svalue/mhvalue fields in textra32)?

3. What happens to the mcontrol.sizehi field if MXLEN is changed from 64 to 32? Can such a change modify the effective access size for which a trigger is activated? For example, if {sizehi:sizelo} is 5 when MXLEN=64 (match on 64-bit accesses), what access size will the trigger match after the XLEN switch?

I'm not saying that any of the cases above are likely or desirable in the real world, but I think it is important to clarify the relationship between the internal state of the system and what is presented by the register view.

pdonahue-ventana, 2020-11-12:
These are good questions.

1. This is answered by section 2.4 of the privileged architecture.  Upon changing to MXLEN=32, textra32[22:18] will be zeroed out.  textra32.mhvalue and textra32.mhselect will take on the values from some upper bits of textra64.svalue.  If MXLEN goes from 32 to 64 then textra64.mhvalue=0 and textra64.mhselect=0 and the previous values of mhvalue and mhselect get written into some upper bits of textra64.svalue.  This is obviously all a bad idea but that's what the architecture says.

2. I may be missing it but I'm not sure how the architecture handles, for instance, M mode writing a CSR that's accessible to S mode but when MXLEN!=SXLEN.  Your question would be answered similarly to however that (more likely) case is handled.

3. I think that CSR width modulation also answers that.

pdonahue-ventana, 2020-11-12:
To solve item 2, I think that all the registers in the trigger module should be defined to be of size MXLEN instead of XLEN.

For instance, in the base architecture all of the supervisor CSRs are defined to be of size SXLEN rather than XLEN (other than a few that are 32b regardless of SXLEN).  If SXLEN=32 then a write of a supervisor CSR from M mode will treat it as a 32b register even if MXLEN=64.

If MXLEN=32 then we would want to use textra32 instead of textra64 even if DXLEN=64 so I think that defining it to be based on MXLEN is what you'd want.  Otherwise, the external debugger might write textra64 but upon leaving debug mode the hardware would interpret it as textra32.

If you later change MXLEN then there are still these weird width modulation effects in items 1 and 3.

timsifive, 2020-11-13:
The procedure described in section 2.4 of the privspec doesn't work at all for our trigger CSRs, because we have type in the upper bits of the register. Additionally, it's not clear how that works if DXLEN=64 and MXLEN=32. The CSRs cannot be converted every time a hart goes between D-Mode and M-Mode. I had hoped, by specifying DXLEN as constant even if MXLEN changes, that we could avoid having to worry about run-time XLEN changes. I guess that's wrong.

For James's scenarios, I had imagined things would work like this:
1. Since the hart does support 64 bits, somewhere it stores 34 bits of svalue. When textra is accessed from a 32-bit mode, it only has access to the low 16 bits of that svalue, and in 32-bit modes triggers will only take into account those low 16 bits. Same with mhvalue.
2. Dynamically reordered.
3. Per my logic above, sizehi would be ignored in a mode where XLEN is 32, even if it was written to something non-zero in a mode where XLEN is 64.

Does that work? What other options do we have to solve this?

JamesKenneyImperas, 2020-11-13:
Hello Tim,

I think your proposal here is basically sound, given the constraint that it is too late to change the register formats to align fields for differing XLEN. As you might have guessed, I'm implementing Trigger Module registers in riscvOVPsim, and the only way I could get it to work was to do almost exactly what you've said here: XLEN=32 and XLEN=64 registers are views onto a common underlying structure that holds all trigger state, with the effect that fields appear at the correct positions automatically in the different-XLEN register views without having to explicitly shuffle them round. This doesn't follow the CSR modulation rules in section 2.4 of the Privileged Specification, but I think those rules are flawed and should be rethought: the same problems we are having here also apply to standard CSRs with variable-width fields by XLEN (e.g. satp).

The one point I would argue with is whether in XLEN=32 mode the upper parts of wider XLEN=64 trigger register fields should be ignored. Developing my third example (size vs sizehi:sizelo), suppose that MXLEN=64, DXLEN=64 and a trigger has been programmed by the Debugger to trap when a 64-bit access is performed (sizehi:sizelo=5). This works as expected while MXLEN=64; however, if the processor switches to MXLEN=32 mode, then ignoring the upper bits of size will mean that the trigger traps on 8-bit accesses instead in that mode, which won't be what is expected. I think it would be better for the triggers to always use the _full width_ of underlying trigger fields, which will usually mean they don't match if inaccessible upper parts of those registers are non-zero, which seems better to me. It also simplifies the trigger implementation somewhat, I think.

James.

closed 2020-12-17 by timsifive